steveicarus/iverilog #481: Combinational assignments don't appear to cascade

jes, 2021-01-18:
## Backgruond

 have ~no idea what I'm doing with Verilog, so it is highly likely that I am just wrong about this. If so, I apologise.

I am trying to design a testbench that I can execute with iverilog to test components designed in Verilog that I will eventually create using TTL chips.

I have a repo here with my test case in: https://github.com/jes/ttl-cpu

## What I did

`alu.v` contains an implementation of the ALU from the Nand2Tetris course, using purely combinational logic via "assign" statements. In my mind this would create a bunch of multiplexers, and an adder and and AND gate, in hardware and a change to an input (e.g. X) would cascade through the multiplexers and the adder and update the output.

You can compile `alu_tb.v` with iverilog (e.g. using the Makefile) and execute the output.

## What I expected to happen

The `a.out` program produced by iverilog would not output anything because the adder would work as intended.

## What happened

It outputs:

    Bad: got 10+10=    0

## Other information

I tried implementing the ALU on a single line, as follows:

    assign out = no?~(f?((nx?~(zx?0:X):(zx?0:X))+(ny?~(zy?0:Y):(zy?0:Y))):((nx?~(zx?0:X):(zx?0:X))&(ny?~(zy?0:Y):(zy?0:Y)))):(f?((nx?~(zx?0:X):(zx?0:X))+(ny?~(zy?0:Y):(zy?0:Y))):((nx?~(zx?0:X):(zx?0:X))&(ny?~(zy?0:Y):(zy?0:Y))));

And then the testbench works as expected. This leads me to believe that the test failure is due to the assignments not cascading the way I expect.

caryr, 2021-01-18:
This is almost certainly not a bug in Icarus. You may want to consider what the width of the intermediate assignment results are versus what you wrote as a single line. That should give you a clue as to why you are getting an incorrect result.

jes, 2021-01-18:
Oh my word...
Thank you! I am a giddy goat. This is indeed not a bug in Verilog, my intermediate assignments were only 1 bit wide...
Thanks a lot.

closed 2021-01-18 by jes