steveicarus/iverilog #482: assigning a 2D array of integer pairs

logicmonkey, 2021-01-18:
I would like to be able to initialise a 2D array of integer pairs with arr[p][q] = {int_a, int_b}.
Is this possible using -g2005-sv ?
Thanks!
`
module intarr();
  integer arr[7:0][3:0][1:0];
  initial begin
    for (int ii=0; ii<8; ii++)
      for (int jj=0; jj<4; jj++) begin
        arr[ii][jj] = {jj, ii};                // error
        // arr[ii][jj][0] = ii; // This...
        // arr[ii][jj][1] = jj; // ...works
      end
  end
endmodule
`

logicmonkey, 2021-01-19:
I get a failure with -g2012 too (which I guess is what I should be selecting anyway if I want SystemVerilog features?)

So aside from the feature above, is there a formal stance on which standard will be best supported going forward? It was mentioned to me somewhere that Icarus SV development had halted due to the size of the task.

caryr, 2021-01-19:
The generation should mostly just control the keywords that are used during parsing and trigger support for SystemVerilog in general.

You have been given bad information regarding our development. SV development is slow in Icarus because we have a small group of developers, but we regularly fix issues and add new features. It was going at a faster pace previously when some additions were being sponsored and we were not as busy with our paying jobs.

This is likely a bug/enhancement in Icarus, but I have not actually checked the code.

logicmonkey, 2021-01-19:
Very glad to hear it - this was pretty much my response, but it's nice to have that confirmed.
Apologies for piggy-backing that onto a ticket.

caryr, 2021-01-21:
I can confirm this is an issue. I believe the issue is we do not currently support assigning unpacked arrays. I'm going to mark this as an enhancement since you have a viable work around.