steveicarus/iverilog #483: $bits on item in a generate block fails when used in a constant expression

zachjs, 2021-02-09:
```SystemVerilog
module top;
	parameter ENABLE = 1;
	if (ENABLE) begin : blk
		wire [7:0] w;
	end
	wire [7:0] x;
	wire [$bits(blk.w)-1:0] y;
	wire [$bits(x)-1:0] z;
	initial begin
		$display("blk.w: %b (%0d bits)", blk.w, $bits(blk.w));
		$display("x: %b (%0d bits)", x, $bits(x));
		$display("y: %b (%0d bits)", y, $bits(y));
		$display("z: %b (%0d bits)", z, $bits(z));
	end
endmodule
```

When simulating the above test case, most tools produce:
```
blk.w: zzzzzzzz (8 bits)
x: zzzzzzzz (8 bits)
y: zzzzzzzz (8 bits)
z: zzzzzzzz (8 bits)
```

However, iverilog (both version 11.0 and the latest master) produces:
```
blk.w: zzzzzzzz (8 bits)
x: zzzzzzzz (8 bits)
y: xx (0 bits)
z: zzzzzzzz (8 bits)
```
Notice that `$bits(blk.w)` correctly evaluates to `8` in the `$display` call, but not in the declaration of `y`.

caryr, 2021-02-09:
I can confirm this is failing with development.

caryr, 2021-02-14:
This are a couple issues around this bug. First generate schemes appear to be elaborated after the enclosing scope. There is also no error reported if the $bits() function is given an invalid/non-existed identifier, the elaboration order is also dependent on the actual name so this can fail even when no generate scope is used. I need to figure out how to trigger an elaboration of the dependent signals to resolve this.

zachjs, 2021-02-21:
I've encountered a distinct but probably related issue on the below case. I'd be happy to file this separately if you'd prefer.
```SystemVerilog
module Module;
    parameter T = 10;
    wire [T-1:0] x;
    initial $display("Module %b %0d", x, T);
endmodule

module top;
    wire [7:0] x;
    Module #($bits(x)) mA();
    Module #(8) mB();
endmodule
```

Most tools yield
```
Module zzzzzzzz 8
Module zzzzzzzz 8
```
whereas iverlog yields
```
Module xx 0
Module zzzzzzzz 8
```

caryr, 2021-02-22:
Yes, that looks like the same issue. I will make sure whatever fix I come up with fixes both.

caryr, 2021-03-07:
I had hoped to get this fixed before the part I have been working on for the last year came back from manufacturing. Unfortunately that did not happen. I'm going to release this for now since I will be lucky if I have time to work on things that actually impact me. If no one else fixes this I will take it up again when I have free time.

martinwhitaker, 2024-02-11:
See discussion #1100.

martinwhitaker, 2024-04-06:
The main issue here is now fixed in the master branch. I'll open a new issue for passing an invalid/non-existent argument to $bits.

closed 2024-04-06 by martinwhitaker

zachjs, 2024-05-08:
Thank you for fixing this issue!