steveicarus/iverilog #485: Default to SystemVerilog 2012

RandyR54, 2021-02-13:
Hi.  Today most designs use some aspects of SystemVerilog, so how about setting the default to SV2012?  An alternative thought would be to add a file that contains default switches.  

Thanks, Randy

caryr, 2021-02-13:
This certainly makes sense. There could be some work needed to make sure ivtest does not break because it is making an assumption regarding the generation. If we decide to switch this I expect it would only be done for development and V12 when it is released.

martinwhitaker, 2021-02-14:
Changing the default would break existing test scripts, so I would discourage that.
If I wanted this, I'd just add an alias, e.g. isv = iverilog -g2012.