steveicarus/iverilog #486: Simple example for generate .dff/p for vvp target?

jimmysitu, 2021-02-22:
Hi, All

I am trying to understand how .dff/p is generated from verilog, so I wrote a simple verilog dff as below
```verilog
module dff(
    input clk, input rst, input d,
    output q
);
reg q;
always@(posedge clk, negedge rst)begin
    if(!rst)begin
        q <= 1'b0;
    end else begin
        q <= d;
    end
end
endmodule
```
But it seem iverilog only translate it to a vthread. It seem I get the wrong idea of .dff/p in vvp code.
Can any provide an example of verilog, which can generate this .dff/p in vvp target?

Thanks

martinwhitaker, 2021-02-22:
Use the iverilog -S option to enable synthesis.

jimmysitu, 2021-02-28:
@martinwhitaker , Thanks, I see .dff/p with -S option.

closed 2021-02-28 by jimmysitu