steveicarus/iverilog #487: accumulator unable to assign to unresolved wires

kgokarn, 2021-02-28:
Hello,

I am using Icarus Verilog for simulations using cocotb testbench. The iverilog version is 11.0 stable. Running on Ubuntu 18.04LTS.

I get an error when I run the simulation on Icarus Verilog "accumulator Unable to assign to unresolved wires". The below code is synthesisable, and also compiles without any error in other compilers such as ModelSim. Please let me know if this case is not handled in Icarus Verilog. Or if there is any workaround.

Thanks in advance.

module macb(
    input wire clk,
    input wire reset,
    input wire next_mac_input,
    input wire signed [11:0] i_activation_0,
    .
    .
    .
    input wire signed [11:0] i_weight_15,
    output wire signed [27:0] o_output,
    output wire ready_out,
    output wire running_out
    );
    
    reg [4:0] state;
    reg ready;
    reg running;
    reg signed [27:0] accumulator;
	reg signed [27:0] acti_weight;
	
	almost_correct_adder aca(.add1_i(accumulator), .add2_i(acti_weight), .result_o(accumulator));

    
   always @ (posedge clk) begin
        if (reset | next_mac_input) begin
            accumulator <= 0;
            ready <= 0;
            state <= 0;
            running <= 0;
        end else begin
            case(state) 
                5'b00000: begin
                    running <= 1;
                    acti_weight <= i_activation_0 * i_weight_0;
                    state <= state + 1;
                end
                .
                .
                .

                5'b01111: begin
					acti_weight <= i_activation_15 * i_weight_15;
                    ready <= 1;
                    running <= 0;
                    state <= state + 1;
                end
                5'b10000: begin
                    state <= 5'b00000;
                    ready <= 0;
                end
            endcase
        end
    end
    
    assign o_output = accumulator;
    assign ready_out = ready;
    assign running_out = running;
    
endmodule

martinwhitaker, 2021-02-28:
Please also provide the code for `almost_correct_adder`. A cut down version will do, providing it shows the error.

kgokarn, 2021-02-28:
Sure. the code of almost_correct_adder is a Verilog gate netlist.

Here is the code:

module almost_correct_adder ( add1_i, add2_i, result_o );
  input [27:0] add1_i;
  input [27:0] add2_i;
  output [27:0] result_o;
  wire   n184, n185, n186, n187, n188, n189, n190, n191, n192, n193, n194,
         n195, n196, n197, n198, n199, n200, n201, n202, n203, n204, n205,
         n206, n207, n208, n209, n210, n211, n212, n213, n214, n215, n216,
         n217, n218, n219, n220, n221, n222, n223, n224, n225, n226, n227,
         n228, n229, n230, n231, n232, n233, n234, n235, n236, n237, n238,
         n239, n240, n241, n242, n243, n244, n245, n246, n247, n248, n249,
         n250, n251, n252, n253, n254, n255, n256, n257, n258, n259, n260,
         n261, n262, n263, n264, n265, n266, n267, n268, n269, n270, n271,
         n272, n273, n274, n275, n276, n277, n278, n279, n280, n281, n282,
         n283, n284, n285, n286, n287, n288, n289, n290, n291, n292, n293,
         n294, n295, n296, n297, n298, n299, n300, n301, n302, n303, n304,
         n305, n306, n307, n308, n309, n310, n311, n312, n313, n314, n315,
         n316, n317, n318, n319, n320, n321, n322, n323, n324, n325, n326,
         n327, n328, n329, n330, n331, n332, n333, n334, n335, n338, n339,
         n340, n341, n342, n343, n344, n345, n346, n347, n348, n349, n350,
         n351, n352, n353, n354, n355, n356, n357, n358, n359, n360, n361,
         n362, n363, n364, n365, n366, n367, n368, n369, n370, n371, n372,
         n373, n374, n375, n376, n377, n378, n379, n380, n381, n382, n383,
         n384, n385, n386, n387, n388, n389, n390, n391, n392, n393, n394,
         n395, n396, n397, n398, n399, n400, n401, n402, n403, n404, n405,
         n406, n407, n408, n409, n410, n411, n412, n413, n414, n415, n416,
         n417, n418;

  xor U323 ( result_o[9], n220, n221);
  xor U324 ( n221, add2_i[9], add1_i[9]);
  xor U325 ( result_o[8], n226, n227);
  xor U326 ( n227, add2_i[8], add1_i[8]);
  xor U327 ( result_o[7], n231, n232);
  xor U328 ( n232, add2_i[7], add1_i[7]);
  xor U329 ( result_o[6], n244, n245);
  xor U330 ( n245, add2_i[6], add1_i[6]);
  xor U331 ( result_o[5], n248, n249);
  xor U332 ( n249, add2_i[5], add1_i[5]);
  xor U333 ( result_o[4], n252, n253);
  xor U334 ( n253, add2_i[4], add1_i[4]);
  xor U335 ( result_o[3], n256, n257);
  xor U336 ( n257, add2_i[3], add1_i[3]);
  xor U337 ( result_o[2], n260, n261);
  xor U338 ( n261, add2_i[2], add1_i[2]);
  xor U339 ( result_o[27], n265, n266);
  xor U340 ( result_o[26], n275, n276);
  xor U341 ( n275, add1_i[26], n213);
  xor U342 ( result_o[25], n280, n281);
  xor U343 ( n281, add2_i[25], add1_i[25]);
  xor U344 ( result_o[24], n295, n296);
  xor U345 ( result_o[23], n302, n303);
  xor U346 ( n303, add2_i[23], add1_i[23]);
  xor U347 ( result_o[22], n313, n314);
  xor U348 ( n314, add2_i[22], add1_i[22]);
  xor U349 ( result_o[21], n322, n323);
  xor U350 ( n323, add2_i[21], add1_i[21]);
  xor U351 ( result_o[20], n326, n327);
  xor U352 ( n327, add2_i[20], add1_i[20]);
  xor U355 ( result_o[19], n338, n339);
  xor U356 ( n339, add2_i[19], add1_i[19]);
  xor U357 ( result_o[18], n348, n349);
  xor U358 ( n349, add2_i[18], add1_i[18]);
  xor U359 ( result_o[17], n352, n353);
  xor U360 ( n353, add2_i[17], add1_i[17]);
  xor U361 ( result_o[16], n360, n361);
  xor U362 ( n361, add2_i[16], add1_i[16]);
  xor U363 ( result_o[15], n370, n371);
[... 198 lines cut ...]
  not U562 ( n194, add1_i[16]);
  not U563 ( n196, add1_i[15]);
  not U564 ( n191, add1_i[18]);
  not U565 ( n189, add1_i[19]);
  nand U566 ( n394, add2_i[10], add1_i[10]);
  nand U567 ( n300, add2_i[22], add1_i[22]);
  not U568 ( n214, add2_i[21]);
  not U569 ( n219, add2_i[13]);
  not U570 ( n217, add2_i[16]);
  not U571 ( n218, add2_i[15]);
  not U572 ( n216, add2_i[18]);
  not U573 ( n215, add2_i[19]);
  nand U574 ( n360, n341, n362);
  nand U575 ( n362, n355, n200);
  not U576 ( n213, add2_i[26]);
  nand U577 ( n226, n228, n229);
  nand U578 ( n229, n230, n224);
  nand U579 ( n252, n254, n255);
  nand U580 ( n254, add2_i[3], add1_i[3]);
  nand U581 ( n255, n256, n211);
  and U582 ( n274, n287, n288, n289);
  nand U583 ( n289, add2_i[24], add1_i[24]);
  nand U584 ( n287, n294, n184);
  nand U585 ( n288, n290, n284);
endmodule

martinwhitaker, 2021-03-01:
A cut-down test case:
````
module dut(output wire result);
  assign result = 1'b1;
endmodule

module top();
  reg value;

  dut dut(value);

  initial value <= 1'b0;
endmodule
````
The error message you report indicates you are compiling for a SystemVerilog generation, so connecting an output port to a variable is allowed. But what isn't allowed is driving the variable from both a procedural assignment and a continuous assignment. From IEEE 1800-2012 section 6.5:

"Variables can be written by one or more procedural statements, including procedural continuous
assignments. The last write determines the value. Alternatively, variables can be written by one continuous
assignment or one port."

I tested this with several other Verilog compilers, and they all reported an error.

kgokarn, 2021-03-01:
Thank you for your reply. For the accumulator, I need to have an initial value (0). Then it must be added with values from the calculation. Therefore, when the hardware is reset, the accumulator will be assigned to 0. How would you change it otherwise?

Also, is changing the accumulator from "reg" to "wire" is suggested?

Thanks in advance.

caryr, 2021-03-02:
Maybe you want to consider that the synthesized hardware is what needs to support the reset instead of trying to implement the reset using the test/procedural code in parallel with the hardware. How would this work if you actually implemented it in hardware which is what the synthesized output is supposed to be? FYI if other simulators support this then as Martin has said they have a bug.

This is not a reg versus wire or Verilog versus SystemVerilog issue. Think about your hardware architecture and solve it that way.

kgokarn, 2021-03-02:
Thank you for your reply. I will consider from the synthesis perspective and rewrite the HDL.  Thanks again.

kgokarn, 2021-03-08:
The issue has been solved. You may close it.

closed 2021-03-09 by caryr