steveicarus/iverilog #488: Verilog preprocessor line directives not always honored

fc-kevin, 2021-03-06:
On both 10.2 (stable) and 11.0 (stable), when you insert preprocessor line directives, they work up until a `` `include `` or multi-line macro is expanded. At that point, it reverts back to the original filename and number.

Line directives are a bit of an esoteric feature, but this is how I understand it. iverilog accepts two flavors of preprocessor line directives,  `` `line `` and `#line`. They both do the same thing (with slightly different syntax) and allow you to rename and renumber the current file being processed for reporting purposes. iverilog itself makes use of these directives, since the preprocessor, ivlpp, inserts them during preprocessing. And they work, to some degree.

Here's an example of a failure condition. Say you have 2 files, `foo.v`:
```verilog
/* This is foo.v:1 */
/* This is foo.v:2 */
`line 10 "hello.v" 0
/* This is foo.v:4, should be reported as hello.v:10 */
`include "bar.v"
/* This is foo.v:6, should be reported as hello.v:12 */
`define MACRO reg A; \
              reg B; \
              reg C;
`line 5 "world.v" 0
/* This is foo.v:11, should be reported as world.v:5 */
module foo;
`MACRO
/* This is foo.v:14, should be reported as world.v:8 */
endmodule
```
and `bar.v`:
```verilog
module bar;
endmodule
```
Here is the output when you run just the preprocessor `/usr/lib/x86_64-linux-gnu/ivl/ivlpp -L foo.v`:
```verilog
`line 1 "foo.v" 0
/* This is foo.v:1 */
/* This is foo.v:2 */
`line 10 "hello.v" 0
/* This is foo.v:4, should be reported as hello.v:10 */

`line 1 "./bar.v" 1
module bar;
endmodule

`line 6 "foo.v" 2   <- BUG
/* This is foo.v:6, should be reported as hello.v:12 */

`line 5 "world.v" 0
/* This is foo.v:11, should be reported as world.v:5 */
module foo;
reg A;
              reg B;
              reg C;

`line 13 "foo.v" 2  <- BUG

/* This is foo.v:14, should be reported as world.v:8 */
endmodule
```

ivlpp inserts `` `line `` directive at the beginning of both files correctly, as it should. But when it comes back from the include, it ends up using the original filename and numbering scheme, not the one that was explicitly added in the file. And again, when coming back from the multi-line macro expansion, it uses the wrong scheme. There may be scenarios where it misbehaves, but these are the ones I've found.

martinwhitaker, 2021-03-13:
This is because the preprocessor does not recognise and handle line directives - they are simply passed through to the main compiler. To properly support line directives in the source files, the preprocessor needs to process them too. This is behaviour specified in the IEEE standard, so we should support it. I'll take a look at doing this.

fc-kevin, 2021-03-14:
Well, that would explain it! Thanks for getting a quick fix in!

I ran the same test as before with 9a2c43e, and the output looks correct. For the record, the results are below.

``` verilog
`line 1 "foo.v" 0
/* This is foo.v:1 */
/* This is foo.v:2 */
`line 10 "hello.v" 0
/* This is foo.v:4, should be reported as hello.v:10 */

`line 1 "./bar.v" 1
module bar;
endmodule

`line 12 "hello.v" 2
/* This is foo.v:6, should be reported as hello.v:12 */

`line 5 "world.v" 0
/* This is foo.v:11, should be reported as world.v:5 */
module foo;
reg A;
              reg B;
              reg C;

`line 7 "world.v" 2

/* This is foo.v:14, should be reported as world.v:8 */
endmodule
```

Reviewing the patch, the only thing that stuck out was the filename parsing. It might not work with weird filenames containing escape characters, in particular escaped double quotes. But that's a separate issue, or perhaps working as intended.

closed 2021-03-14 by fc-kevin