steveicarus/iverilog #489: SystemVerilog feature request: Default port values

nakengelhardt, 2021-03-08:
It seems that icarus verilog does not currently support default port values (as defined in 23.2.2.4 of the SystemVerilog standard), like in the following example:
```
module test (input a = 0, output x); 
  assign x = a; 
endmodule
```
How difficult would this be to add? We have some issues with the Yosys cell simulation library (e.g. https://github.com/YosysHQ/yosys/pull/2390) that would be best solved by using default values, but we want to avoid breaking simulation with iverilog.

martinwhitaker, 2021-03-09:
At first glance this doesn't seem too hard. I'll see what I can do.

martinwhitaker, 2021-03-10:
This is now supported in the master branch. I still need to do a bit more work to restrict it to constant expressions.

martinwhitaker, 2021-03-12:
I've added the restriction that port default values must be constant expressions (as specified in the LRM), so marking this as done.

closed 2021-03-12 by martinwhitaker

nakengelhardt, 2021-03-15:
Awesome, thanks a lot!

meawoppl, 2021-09-06:
Is there a flag you need to pass to support this?

larsclausen, 2022-05-22:
@Forty-Bot Currently only default values on output ports are supported. I have plans for adding support adding support for input default values. But feel free to create a ticket so this doesn't get forgotten.

martinwhitaker, 2022-05-22:
@larsclausen, see earlier comments in this issue. I added support for default values on input ports in March last year.

@Forty-Bot, your example compiles without error. What exactly doesn't work?

larsclausen, 2022-05-22:
@martinwhitaker My bad. I tested it a earlier this year and it did not work. But it looks as if it only does not work for top-level modules. Everywhere else it works.

nachiket, 2025-01-11:
Does this work for top-level ports with v12.0? I can't seem to make this work.

```
$ iverilog -g2012 test.sv

$ ./a.out
x=z,a=z
test.sv:5: $finish called at 2 (1s)

$ cat test.sv
module test (input a, input b = 0, output x);
  assign x = a;
  initial begin
    #1 $monitor("x=%d,a=%d",x,a);
    #1 $finish();
  end
endmodule
```

larsclausen, 2025-01-11:
This is not working for top-level ports. Can you create a new ticket?

larsclausen, 2025-01-12:
yes, thanks.

jmi2k, 2025-03-30:
Also, seems like parameters are not allowed in the RHS of a port definition?

```sv
module Foo #(parameter F) (input logic i = F);
endmodule

module Test;
	Foo #(.F(0)) f();
endmodule
```

```sv
module Foo #(parameter F) (input logic i = F'(1));
endmodule

module Test;
	Foo #(.F(0)) f();
endmodule
```

Those two result in:

```
Icarus Verilog Preprocessor version 12.0 (stable) ()

Copyright (c) 1999-2021 Stephen Williams (steve@icarus.com)

  This program is free software; you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation; either version 2 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License along
  with this program; if not, write to the Free Software Foundation, Inc.,
  51 Franklin Street, Fifth Floor, Boston, MA 02110-1301, USA.
Using language generation: IEEE1800-2012,no-specify,xtypes,icarus-misc
PARSING INPUT
LOCATING TOP-LEVEL MODULES
   Test
 ... done, 0 seconds.
ELABORATING DESIGN
test/sandbox.sv:1: error: Unable to bind parameter `F' in `Test'
test/sandbox.sv:1: error: Failed to elaborate port default value.
2 error(s) during elaboration.
```

However, something like this (which is not equivalent, it's just setting the initial value of a register) works:

```sv
module Foo #(parameter F) (output logic o = F);
endmodule

module Test;
	Foo #(.F(0)) f();
endmodule
```

larsclausen, 2025-03-31:
Yes, please create an issue

jmi2k, 2025-03-31:
Done, #1230