steveicarus/iverilog #490: VPI: cbAtStartofSimTime time value

marlonjames, 2021-03-13:
When registering a `cbAtStartOfSimTime` callback, iverilog interprets the _cb_data_p->time_ field as an absolute simulation time to use, rather than as a delay.

https://github.com/steveicarus/iverilog/blob/45db0db8b3cac64b56dec4b455c0d1e247e05ad7/vvp/vpi_callback.cc#L405-L414

The SV 2017 LRM has these two relevant passages:

>**cbAtStartOfSimTime** Callback shall occur before execution of events in a specified time queue. A
callback can be set for any time, even if no event is present.

>Placing a callback for **cbAtStartOfSimTime** and a delay of zero during a callback for reason
**cbAtStartOfSimTime** will result in another **cbAtStartOfSimTime** callback occurring during the same
time slice.

My interpretation is that the time should be a delay, not an absolute value.

The same wording of "a delay of zero" is used immediately preceding the second passage in context of a **cbReadWriteSynch** callback, which when registered, iverilog interprets the _cb_data_p->time_ field as a delay.

I've also confirmed that Aldec simulators interpret the time as a delay, not an absolute time.

martinwhitaker, 2021-03-13:
I suspect that what led to the current Icarus behaviour was the passage

> cb_data_p->[time->low,time->high,time->real]
> These fields shall contain the requested time of the callback or the delay before the callback.

which implies at least one of the callback types interprets the field as an absolute time, and that there is no other obvious difference beween **cbAtStartOfSimTime** and **cbAfterDelay**. But searching for clarification, I found [https://www.accellera.org/images/eda/sv-bc/0545.html](url), which says

> There is an important difference with **cbAfterDelay** versus **cbAtStartOfSimTime**. The **cbAfterDelay** -- and it counterpart, **tf_setdelay()** -- callback allows a delay of 0, which to the user is like a #0 in the HDL.

So I agree that this is a bug in Icarus, but will wait to see if the other Icarus developers disagree before fixing it.

martinwhitaker, 2021-03-13:
On the other hand, I've just looked at Stuart Sutherland's "The Verilog PLI Handbook", which clearly states that cbAtStartOfSimTime takes an absolute time value.

marlonjames, 2021-03-13:
I found this previous discussion: https://github.com/steveicarus/iverilog/issues/117

The standard is not very clear here, but it's interesting that they use the same wording of "a delay of zero" for **cbAtStartOfSimTime** and **cbReadWriteSynch**, if one uses a delay and one uses an absolute requested time.

It's perfectly possible that Aldec is the odd one out here, though.

martinwhitaker, 2021-03-13:
Yes, I'd forgotten that discussion. My conclusion there stands - the standard is poorly worded. As Stuart Sutherland was one of the leading contributors to the PLI section of IEEE 1364, his book is a good guide to the true intent.

closed 2021-03-13 by marlonjames