steveicarus/iverilog #491: Wrong array adress width

fobekyo, 2021-03-19:
iverilog expects the address of an array to be 2 bits larger than it should be. This can be observed well when converting the following code  to verilog95 using iverilog:

```verilog
module test(input [4:0] addr, output out);
    reg memory [31:0]
    assign out = memory[addr];
endmodule

```
becomes
```verilog
module test(addr, out);
    input [4:0] addr;
    reg memory [31:0];
    wire out;

   assign out = memory[{2'h0, addr} + 0]
endmodule
```

martinwhitaker, 2021-03-21:
Does this cause any real (i.e. behavioural) problems?

There are issues with multi-dimensional arrays - see https://sourceforge.net/p/iverilog/bugs/953/ - but I'm not aware of any with single-dimension arrays.

fobekyo, 2021-03-21:
It creates unnecessary concatenations but it shouldn't cause any behavioral problems.

martinwhitaker, 2021-05-16:
In fact it can cause behavioural problems - see issue 515.

martinwhitaker, 2021-05-16:
Turns out this is not the same bug as the one exposed by issue #515, so goes back to being an annoyance rather than a major issue.