steveicarus/iverilog #493: is there a way to make the simulation run more slowly

Johnlon, 2021-04-02:
I have a sim that interacts with the outside world by way of a UART.
UM245R UART model is https://github.com/Johnlon/spam-1/blob/master/verilog/uart/um245r.v
(no awards for style)

The model isn't the subject of this question - just context.

I want my VVP to run more slowly so that I don't process as many clock cycles between my interactions with the sim.

Otherwise it's running to fast and the logging is furious.

Any ideas?

I guess this is a bit wierd - most folk want more speed probably.

caryr, 2021-04-02:
Can the UART generate an event that can be passed to VVP and then used to trigger an update of the state when it needs to be or maybe you can do something from the VPI (e.g. in the clock generation loop call a VPI routine that waits a passed amount of actual time before returning which should sync the generated clock in Icarus to actual time).

Johnlon, 2021-04-02:
The idea is just to slow the entire sim down vs wallclock time rather than to try to synchonise things.
I only mentioned the Uart for context - I don' really want a solution tied to that.
I want the effect of  running the sim on a very slow PC.

I got the outcome I wanted by hacking the VVP event loop like shown below but I have no idea whether this causes collateral damage.

A configurable delay in the event loop of vvp could be configured by the command line perhaps ...   
   "--loop-slowdown=Xmicrosecs"

`
/*
 * This function runs each thread by fetching an instruction,
 * incrementing the PC, and executing the instruction. The thread may
 * be the head of a list, so each thread is run so far as possible.
 */
void vthread_run(vthread_t thr)
{
      while (thr != 0) {

// STARTPATCH - slow everything down
struct timespec tim, tim2;
tim.tv_sec = 0;
tim.tv_nsec = 10000000;
nanosleep(&tim, &tim2);
// ENDPATCH

        vthread_t tmp = thr->wait_next;
        thr->wait_next = 0;

        assert(thr->is_scheduled);
        thr->is_scheduled = 0;

            running_thread = thr;

        for (;;) {
          vvp_code_t cp = thr->pc;
          thr->pc += 1;

            /* Run the opcode implementation. If the execution of
               the opcode returns false, then the thread is meant to
               be paused, so break out of the loop. */
          bool rc = (cp->opcode)(thr, cp);
          if (rc == false)
            break;
        }

        thr = tmp;
      }
      running_thread = 0;
}`

steveicarus, 2021-04-03:
You can achieve this without patching vvp. Write a vpi module that hooks to
the vpiNetEvent.

caryr, 2021-04-03:
Did you consider the VPI solution I mentioned? That only slows the clock down, but all other processing happens quickly so you should get more predictable delays.

@steveicarus vpiNetEvent? What is that?

Johnlon, 2021-04-03:
I didn't consider the VPI solution as I didn't understand what it meant or how to do it.

Is there a way to grab the wallclock time in iverilog.
If there was then I could write a "sleep(wallclock)" from inside my sim without resorting to writing and linking C code.

Johnlon, 2021-04-03:
Without fully understanding what I was doing I hacked these together that allow me to call the C nanotime lib to put a wallclock sleep into the sim

caryr, 2021-04-03:
Yes, that is what I was suggesting. I'm not certain if you want to do half the delay in each clock phase, but as long as you get something that works that is what really matters. I should have mentioned we implement the system functions using the VPI so the Icarus vpi directory has a huge number of examples.

A couple comments regarding the C code. It's always best to use a compile TF routine to check the argument being passed is of a valid type and the right number of arguments are passed. You can find code for this in the vpi directory. Also in the calltf routine you want to free the argument iterator to avoid a memory leak. Since this is just for you nothing really needs to be done, but these things have a way of getting reused and a little extra time on your part now will make things easier in the future.

If this solved the problem you were facing please close the issue.

Johnlon, 2021-04-03:
Hi thanks.

I've no idea how that argument processing works yet - I just cut that from
something I found on the internet and guessed how to push it into the hello
example from the icarus site.

What's a TF.  Pointers on where to read more beginner stuff appreciated.

Thanks

caryr, 2021-04-04:
You already have it, sleep_compiletf(). Like I said previously, look in the Icarus vpi directory. You likely want to check there is only one integer argument provided. You may want to restrict this to a 32 bit or larger integer constant or variable.

martinwhitaker, 2021-07-31:
This discussion appears to have finished, so closing.

closed 2021-07-31 by martinwhitaker