steveicarus/iverilog #494: Compile fails with GCC-8 static assertion failed: comparison object must be invocable

jmlzone, 2021-04-03:
from some searching around it would appear that some software has issues compiling under gcc-8.

The symptom for icarus Verilog is:
g++ -DHAVE_CONFIG_H -I. -Ilibmisc  -Wall -Wextra -Wshadow   -g -O2 -MD -c PExpr.cc -o PExpr.o
In file included from /usr/include/c++/8/map:60,
                from compiler.h:23,
                from PExpr.cc:25:
/usr/include/c++/8/bits/stl_tree.h: In instantiation of ‘class std::_Rb_tree<enum_type_t*, enum_type_t*, std::_Identity<enum_type_t*>, netenum_t*, std::allocator<enum_type_t*> >’:
/usr/include/c++/8/bits/stl_set.h:133:17:   required from ‘class std::set<enum_type_t*, netenum_t*>’
PExpr.cc:391:40:   required from here
/usr/include/c++/8/bits/stl_tree.h:452:21: error: static assertion failed: comparison object must be invocable with two arguments of key type
      static_assert(__is_invocable<_Compare&, const _Key&, const _Key&>{},
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make: *** [Makefile:249: PExpr.o] Error 1

Noted when trying to compile on Raspberry Pi, but any new linux and GCC-8 should see the issue.

martinwhitaker, 2021-04-03:
Not reproducible here using either GCC 8.4.0 or 10.2.1 on Linux. What version of GCC do you have? If it's older than 8.4, this may be https://gcc.gnu.org/bugzilla/show_bug.cgi?id=85965.

jmlzone, 2021-04-03:
Hi I am using GCC 8.3.0 on raspbian. (8.3.0-6+rpi1)

I will see if there is a gcc 8.4 for raspbian.

thanks,
James

jmlzone, 2021-04-06:
I have upgraded to gcc10.1 compiles and works.  I guess it's a gcc 8.3 bug.

closed 2021-04-06 by jmlzone