steveicarus/iverilog #496: vpiPackage returned when iterating over vpiModules

toddstrader, 2021-04-13:
For reference:
https://github.com/verilator/verilator/issues/2887

If you run `t_vpi_module.pl --iv` on HEAD of Verilator you'll get this:
```
%Error: t_vpi_module.cpp:95: GOT = 600   EXP = 32
%Error: t_vpi_module.cpp:95: C Test failed
```
which tells us that even though we asked to vpi_iterate() over vpiModules (32) we got back a vpiPackage (600).  I believe this is wrong because 38.23 says:
> The vpi_iterate() routine shall return a handle to an iterator, whose type shall be
vpiIterator, which can used by vpi_scan() to traverse all objects of type type associated with object ref.

Which to me reads like it can **only** return objects of type `type`.  Thoughts?  Am I reading this correctly?

I'm currently running iverilog 10.1.

martinwhitaker, 2021-04-13:
You are reading it correctly, and it's a bug in vvp. It assumes all objects at the root of the hierarchy are modules. That was a valid assumption before SystemVerilog came along...

BTW v10.1 is 5 years old. The v10 branch is no longer maintained.

martinwhitaker, 2021-04-18:
Fixed in both the master and v11 branches.

closed 2021-04-18 by martinwhitaker