steveicarus/iverilog #502: SDF Error with IOPATH

hblxuor, 2021-04-22:
I'm trying to compile verilog with an SDF file. iverilog cannot seem to find the ports associated with the IOPATH calls.  iverilog compiles the verilog file correctly when the SDF file is not included.

Here is the relevant patch of SDF:
  (CELL
     (CELLTYPE "latchdr")
     (INSTANCE g3970_L1)
     (DELAY
        (ABSOLUTE
          (IOPATH D Q (::148) (::148))
          (IOPATH ENA Q (::147) (::147))
          (IOPATH R Q () (::110))
        )
     )
     (TIMINGCHECK
        (RECREM (posedge R) (negedge ENA) (::80) ())
        (SETUPHOLD (negedge D) (negedge ENA) (::100) (::100))
        (SETUPHOLD (posedge D) (negedge ENA) (::100) (::100))
     )
  )

The relevant lines in the verilog file:
  latchdr g3970_L1(.R (n_340), .ENA (n_339), .D (n_338), .Q (n_337));
...
module latchdr(ENA, D, R, Q);
  input ENA, D, R;
  output Q;
  wire ENA, D, R;
  wire Q;
  wire next_state;
  reg  qi;
  assign #1 Q = qi;
  assign next_state = D;
  always
    @(next_state or ENA or R)
      if (!R)
        qi <= 0;
      else begin
        if (ENA)
          qi <= next_state;
      end
  initial
    qi <= 1'b0;
endmodule

And the error messages:
SDF WARNING: teststmp.v:157: Unable to match ModPath D -> Q in testbench.dut.g3970_L1
SDF WARNING: teststmp.v:157: Unable to match ModPath ENA -> Q in testbench.dut.g3970_L1
SDF WARNING: teststmp.v:157: Unable to match ModPath R -> Q in testbench.dut.g3970_L1

I get one of these for every element in the SDF.

martinwhitaker, 2021-04-23:
You need to declare the module paths in a specify block. When you have done that, compile with the `-gspecify` option. Note that timing checks are not implemented and will be ignored.

hblxuor, 2021-04-23:
Thanks! This is my first time using SDFs so I've been stumbling around quite a bit.

closed 2021-04-23 by hblxuor