steveicarus/iverilog #503: vpip_set_callback workaround causes issues on Windows with multiply DLLs linkage

Shatur95, 2021-04-23:
When running on Windows, I get the following message:

```
Assertion failed: vpip_routines, file libvpi.c, line 42
```

This happens when I create a DLL with basic VPI functionality (not an iverilog plugin), and then create a plugin for Iverilog, linking the above DLL into the plugin. The problem is caused by this line:

https://github.com/steveicarus/iverilog/blob/a7cb93842eabf2e820887785702defb4e1ebf66a/vpi/libvpi.c#L25

Since `libvpi` is a static library, when linking it to different dynamic libraries (base DLL and iverilog plugin), I get several instances of this variable, but in this line only one such variable will be set:

https://github.com/steveicarus/iverilog/blob/a7cb93842eabf2e820887785702defb4e1ebf66a/vvp/vpi_modules.cc#L223

Is it possible to call `vpip_set_callback` on each linked library or use another approach without jumptable on Windows?

martinwhitaker, 2021-04-24:
Try adding an empty vlog_startup_routines table to your base DLL and instructing vvp to load that module first.

If you can find a solution to allowing multiple programs to use the same DLL that doesn't need a jump table, I'd be happy to use it.

martinwhitaker, 2021-07-31:
No further feedback, so closing. If you have any suggestions for improving the way it works on Windows, please do let us know.

closed 2021-07-31 by martinwhitaker