steveicarus/iverilog #504: Can't call void function from some functions/tasks

udog14, 2021-04-30:
I got a assert error during compilation when a void function is called from some certain tasks and functions in my code.  I can't generalize where this happens but here's a small example.

module test;

  initial begin
    v1;
    v2;
    $finish;
  end

  function void v1;
    $display("v1 is called");
    v2;  // this is not OK
  endfunction

  function void v2;
    $display("v2 is called");
    //v1;  // this is OK
  endfunction

endmodule

This is the error message from running "iverilog -g2012 *.v":

> void_func.v:11: assert: elaborate.cc:3838: failed assertion dscope->elab_stage() >= 3
> sh: line 1: 3905250 Done                    /usr/local/lib/ivl/ivlpp -L -F"/tmp/ivrlg25058eb2e" -f"/tmp/ivrlg5058eb2e" -p"/tmp/ivrli5058eb2e"
>      3905251 Aborted                 (core dumped) | /usr/local/lib/ivl/ivl -C"/tmp/ivrlh5058eb2e" -C"/usr/local/lib/ivl/vvp.conf"

So, v1 can't call v2.  However, v2 can call v1 fine (comment line 11 and uncomment line 16).
Also, If you remove 'void' from v2, it will compile OK, just with a warning that v2 is being called as a task.

martinwhitaker, 2021-05-01:
This compiles and runs without error for me, using both the head of the master branch and the head of the v11 branch. What version of iverilog are you using?

udog14, 2021-05-02:
Ah, the error was from "11.0 (stable) (v11_0)".  Looks like that was the version bundled in Qflow.  I thought I was already trying the head of master but it turned out I got mixed up.   This is no problem then.  Thanks.

closed 2021-05-02 by udog14