steveicarus/iverilog #508: Unexpected infinite loop

pniavis, 2021-05-05:
I can not simulate my design because iverilog gets stuck in an infinite loop. The following code demonstrates the problem.

```systemverilog
module cpu();

logic [1:0] dout, data, ir;
logic rd;

initial begin
    #1;
    $finish();
end

always_comb begin : B
    $display("B");
    dout = 2'b01;
    dout[1] = dout[0];
end

always_comb begin : A
    $display("A");
    ir = 2'h0;
    ir = 2'h1;
    rd = ir[1];
    data = dout;
end

endmodule
```

I would expect to see either "BA" or "ABA" printed and then the simulation finish. However I am seeing ABABABA..... If I exchange the order of the A and B always_comb blocks in the module, then the simulation prints BA and finishes.

I compile the file using the following command
```sh
iverilog -g2012 -pfileline=1 -v  cpu.sv
```

Modelsim can run the code without issues. Have I hit some undefined behavior or it is a bug?

martinwhitaker, 2021-05-16:
Yes, this is a bug in the implementation of always_comb.

martinwhitaker, 2021-05-18:
Fixed in both the master and v11 branches.

closed 2021-05-18 by martinwhitaker