steveicarus/iverilog #510: What causes Thread words exhausted?

mickey4u, 2021-05-11:
I want to find out what causes the error `vvp.tgt error: Thread words exhausted.` to be thrown.

martinwhitaker, 2021-05-16:
Well, your example fails to compile because `product` is not defined. But the obvious answer is that the RHS expression in the assignment to `out` is too complex. Thread words are used for temporarily holding things like index values and shift distances. If you have nested sub-expressions, a thread word allocated at an outer level will not be freed until that level has been fully evaluated.

mickey4u, 2021-05-16:
@martinwhitaker Am I right to say a thread is created for every sub-expression and each thread waits for it's internal sub-expression to finish evaluation; Icarus has a limit on the number of these threads that are spawned. Also, what is the supported  maximum depth of expression in Icarus Verilog.

steveicarus, 2021-05-16:
Icarus verilog doesn't create threads in the sense that you commonly know,
but instead creates objects that represent the thread in the Verilog sense.
And those objects use stacks for expression evaluations. The exceptions are
that there are some index "registers" that are used for index operations
and a few other special cases. That's probably what's running out for you.

martinwhitaker, 2021-07-31:
This discussion appears to have finished, so closing.

closed 2021-07-31 by martinwhitaker