steveicarus/iverilog #512: Releases frequency

mciepluc, 2021-05-12:
Hello,

Would it be possible to make releases more frequently?
I know that there is an option installing from bare sources, but having release with tagged version is much more convenient.
Especially in environments when each update requires reinstallation.

Thanks.

vborchsh, 2022-02-12:
Let me bump this request.

Example: I have CI in the gitlab. Unfortunately, every time I have to build Icarus from sources to be sure, that I use the most updated version. It might take a while and it spend my CI minutes to nothing.

Is there a chance to get a per-week releases, at least? Something like the "nightly" builds?

Thanks.

Luflosi, 2023-04-18:
The latest yosys release is not fully compatible with the latest release of iverilog: https://github.com/YosysHQ/yosys/issues/3731.
Please create a new release so distributions don't need to ship iverilog from Git.

larsclausen, 2023-04-18:
The latest release is v12.0, which was released at the end of December 2022. See https://github.com/steveicarus/iverilog/releases/tag/v12_0

The linked yosys code works with the v12.0 release.

Luflosi, 2023-04-18:
Thanks for the information.
I didn't see the release since the tag does not have an accompanying GitHub release and I didn't check the list of tags.

caryr, 2023-06-11:
I have add v12.0 as the latest release on both github and sourceforge. It looks like that was missed when the tag was created. Do we still need to keep this open?

Luflosi, 2023-06-11:
IMO this can be closed. @mciepluc any opinions?

mciepluc, 2023-06-12:
Well, I asked if it is possible to make more frequent releases. We have 2 years between v 11 and 12. I can see a lot is happening in the project - new official release every few months would be much appreciated.

caryr, 2023-06-12:
I doubt we will ever get to a release every few months. I think a reasonable expectation is minor releases once or twice a year and major releases every year or so. Would development snapshots help?

martinwhitaker, 2024-01-08:
Closing as answered - even if the answer wasn't what you were hoping for!

closed 2024-01-08 by martinwhitaker