steveicarus/iverilog #514: Unable to compile function defined inside package

treforsouthwell, 2021-05-12:
verilog -g2012 test_pkg_.sv 
test_pkg_.sv:3: syntax error
test_pkg_.sv:2: error: Syntax error defining function.

package test_pkg_;
   function automatic int rgx_usc2_itr_alu_gateway;
   input  int                                           sample_count ;
   endfunction // test_pkg_
endpackage

martinwhitaker, 2021-05-16:
This is because the compiler doesn't currently support using the extended SystemVerilog types in the first form of task or function declaration syntax. You can work round this by using the second form, e.g.
````
package test_pkg_;
  function automatic int rgx_usc2_itr_alu_gateway(input int sample_count);
  endfunction
endpackage
````

larsclausen, 2022-04-23:
This has been implemented in https://github.com/steveicarus/iverilog/pull/623. Ticket can be closed.

closed 2022-04-24 by steveicarus