steveicarus/iverilog #515: buffer pointer does not roll back to zero

aniltirli, 2021-05-13:
I wrote a simplified version of the buffer as described below: 
takes 1 byte inputs and outputs 3 bytes, 
stores 8 bytes of data

I expect buffer to wrap around and output last, first and second bytes after array index rdPntr+1 and rdPntr+2 overflow(ie. o_data[23:0] = "070001"). **Verilator** and **simulator of vivado** give the expected output. However, **iverilog** displays "07xxxx" in wave trace as i attached below.

code: 
```
module lineBuffer(
input   i_clk,
input   i_rst,
input [7:0] i_data,
input   i_data_valid,
output [23:0] o_data,
input i_rd_data
);

reg [7:0] line [7:0]; //line buffer
reg [2:0] wrPntr;
reg [2:0] rdPntr;

always @(posedge i_clk)
begin
    if(i_data_valid)
        line[wrPntr] <= i_data;
end

always @(posedge i_clk)
begin
    if(i_rst)
        wrPntr <= 'd0;
    else if(i_data_valid)
        wrPntr <= wrPntr + 'd1;
end

assign o_data = {line[rdPntr],line[rdPntr+1],line[rdPntr+2]};

always @(posedge i_clk)
begin
    if(i_rst)
        rdPntr <= 'd0;
    else if(i_rd_data)
        rdPntr <= rdPntr + 'd1;
end

endmodule
```
iverilog trace:
[image]

verilator trace:
[image]

logicmonkey, 2021-05-13:
If I had a test bench for that code I would try adding modulus 8 to those indices:
`assign o_data = {line[rdPntr],line[(rdPntr+1)%8],line[(rdPntr+2)%8]};`
The simulator will be using a pointer to those array elements rather than a 3 bit binary value, and that pointer has probably read beyond the array bounds and returned Xes.
In RTL you occasionally have to enforce your design intent rather than assume the tool knows what you meant.
I'm not an iverilog developer, just a user. I hope that fixes it - it may not!

martinwhitaker, 2021-05-16:
You need to learn how Verilog determines expression bit lengths. This is described in the Verilog LRM, e.g. IEEE 1364-2005 section 5.4. The TL;DR version is that the expression `rdPntr+1` has a bit length of 32 bits (because it contains an unsized number) so will not wrap round until you reach 2^32.

The Verilog LRM states that an out-of-bounds array access returns X's.

Verilator does not support 4-state values, but I would have expected it to output 0's for out-of-bounds array accesses - but maybe it optimises for speed rather than for correctness. All other simulators I've tried give the same output as Icarus.

It should be possible to get the result you expected by explicitly sizing the constant values, i.e.
````
 assign o_data = {line[rdPntr],line[rdPntr+3'd1],line[rdPntr+3'd2]};
````
Unfortunately array index expression sizing got broken between v10 and v11 (see issue 491) and this no longer works in Icarus. To work around that, either do as @logicmonkey suggests, or use intermediate variables to hold the other index values.

Of course most synthesis tools will simply discard the unused bits of the index expression and give the result you expected.

martinwhitaker, 2021-05-16:
I've pushed a fix for the regression mentioned above to both the master and v11 branches, so
````
 assign o_data = {line[rdPntr],line[rdPntr+3'd1],line[rdPntr+3'd2]};
````
now works as it should.

closed 2021-05-16 by martinwhitaker