steveicarus/iverilog #516: Dumped verilog with -P option gives default to all case expressions.

minu990122, 2021-05-17:
Original Verilog
```
  1 module test (
  2         clk
  3         );
  4
  5 input clk;
  6
  7 reg a;
  8
  9 always @ (posedge clk) begin
 10     case (a)
 11         0:
 12             begin
 13                 a <= 1;
 14             end
 15         1:
 16             begin
 17                 a <= 0;
 18             end
 19         default:
 20             begin
 21                 a <= 0;
 22             end
 23     endcase
 24 end
 25
 26 endmodule
```

Dumped Verilog
```
  1 PFORM DUMP NATURES:
  2 PFORM DUMP DISCIPLINES:
  3 PFORM DUMP COMPILATION UNITS:
  4 package $unit
  5 endpackage
  6 PFORM DUMP PACKAGES:
  7 PFORM DUMP MODULES:
  8 module test;
  9     .clk(clk)
 10     reg logic scalar set_data_type_=13vector_type_t net<scalar> a;
 11         vector of logic
 12     wire /*implicit*/ input <no_type> scalar port<scalar> clk;
 13     always /* case.v:9 */
 14       @(posedge clk)
 15         begin
 16           case (a) /* case.v:10 */
 17             default:
 18                 begin
 19                   a <= 'sd1;  /* case.v:13 */
 20                 end
 21             default:
 22                 begin
 23                   a <= 'sd0;  /* case.v:17 */
 24                 end
 25             0:
 26                 begin
 27                   a <= 'sd0;  /* case.v:21 */
 28                 end
 29           endcase
 30         end
 31 endmodule
 32 PFORM DUMP PRIMITIVES:
```

In pform_dump.cc, line 938 dumping PCase, ```"default:"``` is dumped when ```cur->expr``` is not empty.
It seems to dump ```"default:"``` when ```cur->expr``` is empty.

After changing above, it dumps same memory address twice.
In line 943 and 946, dumping ```*idx_exp``` is ```PExpr*```.
It should use ```(*idx_exp)->dump(out);``` to dump ```expr```.

Between line 943 and 946, ```idx_exp``` doesn't increase.
It should add ```idx_exp++;``` between two lines, not to dump the same ```expr``` twice.

I changed the code as below and it seems to dump case statements O.K.

Original code
```
// line 934 in pform_dump.cc
    for (unsigned idx = 0 ;  idx < items_->count() ;  idx += 1) {
        PCase::Item*cur = (*items_)[idx];

        if (! cur->expr.empty()) { // line 938
          out << setw(ind+2) << "" << "default:";

        } else {
          list<PExpr*>::iterator idx_exp = cur->expr.begin();
          out << setw(ind+2) << "" << *idx_exp; // line 943

          for( ; idx_exp != cur->expr.end() ; ++idx_exp) {
            out << ", " << *idx_exp; // line 946
          }

          out << ":";
        }

        if (cur->stat) {
          out << endl;
          cur->stat->dump(out, ind+6);
        } else {
          out << " ;" << endl;
        }
      }
line 958
```

Changed version
```
// line 934 in pform_dump.cc
    for (unsigned idx = 0 ;  idx < items_->count() ;  idx += 1) {
        PCase::Item*cur = (*items_)[idx];

        if (cur->expr.empty()) { // line 938 removed !
          out << setw(ind+2) << "" << "default:";

        } else {
          list<PExpr*>::iterator idx_exp = cur->expr.begin();
          out << setw(ind+2) << "";
          (*idx_exp)->dump(out); // line 943 dump expr
          idx_exp++; // line 944 move to next expr
          for( ; idx_exp != cur->expr.end() ; ++idx_exp) {
            out << ", ";
            (*idx_exp)->dump(out); // line 946 dump expr
          }

          out << ":";
        }

        if (cur->stat) {
          out << endl;
          cur->stat->dump(out, ind+6);
        } else {
          out << " ;" << endl;
        }
      }
// line 958
```

minu990122, 2021-05-17:
BTW I used ```./ivl -P dump.v case.v``` to produce.

caryr, 2021-06-21:
Thank you for the report and fix. I have updated both V11 and development.

closed 2021-06-21 by caryr