steveicarus/iverilog #517: `recv_real_from_inputs` is unimplemented: would appreciate some guidance as to how to work around it

donn, 2021-05-24:
Issue lies here: https://github.com/steveicarus/iverilog/blob/1f8876be1c49a43e6d8680461a531604ad989388/vvp/vvp_net.cc#L3440

So it's just plain unimplemented, which is fine, but there appears to be no guidance as to what triggers this function call or if there are any workarounds. Can anyone weigh in?

martinwhitaker, 2021-05-24:
`recv_real_from_inputs` is an optional method implemented by subclasses of `vvp_wide_fun_core`. There are only three such classes, `sfunc_core`, `ufunc_core`, and `vvp_udp_fun_core`. `sfunc_core` and `ufunc_core` implement `recv_real_from_inputs`, so it would seem this is coming from an instance of `vvp_udp_fun_core`. That would represent a UDP in your netlist. UDP ports are single bits, so would not expect to receive a real value, and the compiler really shouldn't generate code that causes that to happen.

If you can produce a cut-down example that demonstrates the problem, we can investigate further.

donn, 2021-05-25:
Thanks for replying-  I think I might be able to, but it will take some time. There's a proprietary library in the mix, unfortunately, so I have to go in and isolate the problem without leaking any code. I'll keep you apprised.

closed 2021-11-07 by donn