steveicarus/iverilog #520: How to include Xilinx library into Iverilog? (Xilinx IP core)

xiuwen8686, 2021-06-02:
I want to simulate the design that includes the Xilinx IP with Iverilog.
 I  included the lib by " iverilog -y %vivado_dir%/unisims. ",but only one IP core  can be used.

The Xilinx document mentions the need to compile the library(unisims).
Icarus verilog isn't in the available list of simulators for the Xilinx's command  "compile_simlib"  -simulator switch.

Is there any other way to compile the lib?

martinwhitaker, 2021-06-02:
Icarus Verilog does not support separate compilation, so there is no need to compile the library in advance.

Running on Linux, with the Xilinx software installed in /opt/xilinx, the following example
````
module test();
  reg  clk, set, rst, ena, d1;
  wire d2, d3;

  FDSE ff1(.C(clk), .CE(ena), .S(set), .D(d1), .Q(d2));
  FDCE ff2(.C(clk), .CE(ena), .CLR(rst), .D(d1), .Q(d3));
endmodule
````
can be compiled by
````
iverilog  -y /opt/xilinx/Vivado/2020.2/data/verilog/src/unisims /opt/xilinx/Vivado/2020.2/data/verilog/src/glbl.v test.v
````

xiuwen8686, 2021-06-03:
I have two IP core : Block memory generator and multiplier.
When I use this command to compile my project, multiplier will report some errors.
[image]
Could you give me a few pointers with this problem? Thank you.

martinwhitaker, 2021-06-03:
The compiler is reporting a syntax error on line 185 of IP/mult_gen_1.v. So you need to look at that line (and possibly the surrounding lines) to see what is unusual about it. If it is using any SystemVerilog features, add the `-g2012` option to the iverilog command line. If it contains attributes (enclosed in `(* ... *)`), try temporarily changing them into comments.

xiuwen8686, 2021-06-03:
Add the `-g2012` option to the iverilog can solve the problem. I appreciate your help very much
But the syntax error on line 185 of IP/mult_gen_1.v 
[image]
I can't solve the error. Could you give me a few pointers with this problem?  Thank you.

steveicarus, 2021-06-04:
Icarus Verilog doesn't support encrypted IP.

xiuwen8686, 2021-06-04:
> Icarus Verilog doesn't support encrypted IP.
Oh,I see.   I appreciate your help very much.

martinwhitaker, 2021-06-04:
An open source simulator can't support encrypted IP, because anyone could modify the simulator to output the decrypted code. So closing, as there is no way we can support this.

closed 2021-06-04 by martinwhitaker