steveicarus/iverilog #521: Multi-dimensional packed array with for-loop index

udog14, 2021-06-07:
The code below sets bit 3 of all members of array a.  I should print '8888' (tested with vcs).
However, I got this error running with "iverilog -g2012"

> mda.v:8: error: A reference to a wire or reg (`i') is not allowed in a constant expression.
> mda.v:8: error: Array index expressions must be constant here.

`
module test;

  logic [3:0][3:0] a;

  initial begin
    a = 0;
    for (int i=0; i<4; i++)
      a[i][3] = 1;
    $display("%h", a);
    $finish;
  end

endmodule
`

martinwhitaker, 2021-07-27:
According to this comment in the code:
````
/*
 * Evaluate the prefix indices. All but the final index in a
 * chain of indices must be a single value and must evaluate
 * to constants at compile time. For example:
 *    [x]          - OK
 *    [1][2][x]    - OK
 *    [1][x:y]     - OK
 *    [2:0][x]     - BAD
 *    [y][x]       - BAD
 * Leave the last index for special handling.
 */
````
this was a deliberate restriction, but I can't find any justification for it in the IEEE standard. Maybe Steve can remember why, although it was written 9 years ago, so maybe not...

jgaztelu, 2022-04-18:
Hi, I have encountered the same issue when trying to verify some existing code. 
It looks like the issue with the outer prefixes happens only inside of always blocks, and not when using for .. generate loops (I assume these are handled by a completely different part of the code). 

So, in my experience, this works:

``` Systemverilog
 generate
        for (genvar i=0; i<par_lines_p; i++) begin
            assign r_bit[i] = i_red[i][i_fetch_bit];
            assign g_bit[i] = i_green[i][i_fetch_bit];
            assign b_bit[i] = i_blue[i][i_fetch_bit];
            assign red_row_sh[i] = {red_row[i][width_p-2:0],r_bit[i]};
        end
 endgenerate
```
While this doesn't:
``` Systemverilog
always_ff @(posedge clk) begin : prefetch_proc
      red_row[i] <= {red_row[i][width_p-2:0],r_bit[i]};
end
```
I can confirm this code works with the Vivado simulator and AFAIK it should be legal Systemverilog.

Anyway, not sure if it helps much but it might be helpful if somebody stumble upon the same issue 😊

udog14, 2022-04-18:
I've forgotten this.
@martinwhitaker I agree that this would be illegal
 a[2:0][x] = ...
because the bus is not contiguous (and thus not really a bus).

But a[y][x] is definitely legal.

jmlzone, 2023-11-28:
Seeing a similar issue but with structures.  This notation works in other simulators.
I just did a git pull and rebuilt to ensure I was up to date:
Icarus Verilog version 13.0 (devel) (s20221226-331-g77d7f0b8f)

iverilog -I ../rtl -g2012 -o bist.vvp -c bist.vc
../rtl/marchc_mbist.sv:81: sorry: constant selects in always_* processes are not currently supported (all bits will be included).
../rtl/marchc_mbist.sv:195: error: A reference to a net or variable (`r') is not allowed in a constant expression.
../rtl/marchc_mbist.sv:195: error: Array index expressions must be constant here.
ivl: netmisc.cc:1619: NetExpr* collapse_array_indices(Design*, NetScope*, NetNet*, const std::__cxx11::list<index_component_t>&): Assertion `rc' failed.
Aborted

The For loop on line 189:
      for(int r=0; r<NUM_RAMS; r++)
is unrolled and synthesizable and simulatable.

On 195 I reference into the structure. 
Full RTL below.

/*--------------------------------------------------------------------------------
 (C) Copyright 2004-2023 James M. Lee
----------------------------------------------------------
$Id:$
$Source$
----------------------------------------------------------
A perpetual non-exclusive license to use is hereby granted
to any consulting customer of James M. Lee
----------------------------------------------------------
Revision History
----------------------------------------------------------
Date   Who                Description
----------------------------------------------------------
 8/26/2004 James Lee  Initial Creation
 1/12/2007 James Lee  Adaptation and simplification
 10/20/2023 James Lee Move into SystemVerilog package
--------------------------------------------------------
 
 a simple march memory bist controller

 ref:  
 https://www.technoarete.org/common_abstract/pdf/IJERECE/v4/i10/Ext_98031.pdf
 patterns 
  1) write all zeros to all memories   incrementing the address. (no read)
  2) read back the zeros and write 1's incrementing the address.
  3) read back the ones and write 0's  incrementing the address.
  4) read back the zeros and write 1's decrementing the address.
  5) read back the ones and write 0's  decrementing the address.
  6) read back the zeros               decrementing the address. (no write)
--------------------------------------------------------------------------------*/
import tap_pkg::*;
module march_mbist #(parameter MAX_ADDR=4095, DATA_WIDTH=32, PIPE_DEPTH=1, NUM_RAMS=1, ADDR_BITS=$clog2(MAX_ADDR)-1 ) (
  input clock,
  input reset_n,
  input start,
  input [DATA_WIDTH-1:0] dataout [0:(NUM_RAMS-1)],
  output logic we_n,
  output logic [ADDR_BITS:0] addr,
  output logic [DATA_WIDTH-1:0] datain,
  output logic running,
  output logic fail
 );

typedef struct packed {
  logic check;
  logic expval;
  logic [ADDR_BITS:0] addr;
} CHECK_T;

typedef struct packed {
  logic failed;
  logic [DATA_WIDTH-1:0] bits;
[... 131 lines cut ...]
    if((phase == IDLE) & start)
      begin
	fail_nxt = '0;
      end // if ((phase == IDLE) & start)
    else 
      for(int r=0; r<NUM_RAMS; r++)
	begin
	  ram_dout = dataout[r];//[r];
	  already_failed = fail_status[r].failed;
 	  if(~already_failed & check & (ram_dout != expect_data)) 
	    begin
	      fail_nxt[r].failed = 1'b1;
	      fail_nxt[r].addr = check_pipe[PIPE_DEPTH].addr;
	      fail_nxt[r].bits = ram_dout ^ expect_data;
	    end // if (~already_failed & check & (ram_dout !=expect_data))
	  else
	    begin
	      fail_nxt[r] = fail_status[r];
	    end // else: !if(~already_failed & check & (ram_dout !=expect_data))
	end
    fail=0;
      for(int f=0; f<NUM_RAMS; f++)
	fail = fail | fail_status[f].failed;
  end
endmodule // march_mbist

jhallen, 2024-02-08:
I wish this was fixed..

>this was a deliberate restriction, but I can't find any justification for it in the IEEE standard. Maybe Steve can remember why, although it was written 9 years ago, so maybe not...

I remember commercial simulators had similar restrictions: for example you can't have vec[j+7:j] (maybe too hard to prove that the slice width is always constant), so they added vec[j :+ 8].  If this can work, I think it's reasonable to expect var[j][7:0] to also work.

If it's on the right side, you can use a temporary variable as a work-around: tmp = vec[j]; tmp[7]...

cousteaulecommandant, 2024-04-22:
For the record, this issue only happens on multidimensional *packed* arrays.  You can do `x[i][j][k]` just fine if the first 2 dimensions of `x` are unpacked.
____
> I remember commercial simulators had similar restrictions: for example you can't have vec[j+7:j] (maybe too hard to prove that the slice width is always constant), so they added vec[j :+ 8].

I understand that this is a limitation of standard Verilog.  A range expression is either `constant_expr : constant_expr` or `expr +: constant_expr` (or `-:`), but it cannot be `expr : expr` with two variable expressions (which makes sense because that'd allow variable width ranges).  This applies even to the case where both expressions are a fixed value apart (e.g. `j+7 : j`), probably because it's hard to determine that at compile time.
In any case, I don't think this bug is related to ranges.