steveicarus/iverilog #523: About the keyword '.concat' in the VVP file

slivebullet, 2021-06-12:
I found the keyword'.concat' in the VVP file. I have a few questions and I hope to get your answers. I also read the README of the VVP emulator source code on GitHub, but I still have a few doubts. If you are willing to help me answer, I will be grateful.
### Question 
What is the meaning of the keyword ".concat"? Is there any connection with the "assign" keyword in Verilog？

### Question 
The following quote is from the README in the source code, I have a few things that I don’t understand.
>The opposite of the part select statement is the part concatenation
statement. <u>The .concat statement is a functor node that takes at input</u>
<u>vector values and produces a single vector output that is the</u>
<u>concatenation of all the inputs.</u>
        	<label> .concat [W X Y Z], <symbols_list> ;
*The "[" and "]" tokens surround a set of 4 numbers that are the*
*expected widths of all the inputs. These widths are needed to figure*
*the positions of the input vectors in the generated output, and are*
*listed in order LSB to MSB. The inputs themselves are also listed LSB*
*to MSB, with the LSB vector input coming through port-0 of the real*
*functor.*
The initial output value is (W+X+Y+Z) bits of 'bx. As input values are
propagated, the bits are placed in the correct place in the output
vector value, and a new output value is propagated.

1. The first place is underlined, I am puzzled. Does it mean that ‘.concat’ is a connection point, similar to a connection point of multiple input and single output, which is similar to the USB hub in our lives? If so, what is the relationship between output and input?
2. I am also confused by the italicized part of the text，Do the four numbers in square brackets represent the expected width of the input? If yes, how to explain the VVP source code below?The four numbers in the square brackets are 1,5,0,0; which number in the square brackets represents the expected width of `L_000001a7e8b827e0`?
3. >These widths are needed to figure the positions of the input vectors in the generated output

I don’t really understand the meaning of this sentence. What does the position of the input in the output mean?

```assembly
v000001a7e8b53770_0 .net *"_ivl_0", 0 0, L_000001a7e8b827e0;  1 drivers
L_000001a7e8bc7838 .functor BUFT 1, C4<00000>, C4<0>, C4<0>, C4<0>;
v000001a7e899bb60_0 .net *"_ivl_5", 4 0, L_000001a7e8bc7838;  1 drivers
v000001a7e8b53330_0 .net "a", 5 0, v000001a7e8b860a0_0;  1 drivers
v000001a7e8b85f60_0 .net "b", 5 0, v000001a7e8b86140_0;  1 drivers
v000001a7e8b86000_0 .net "c", 5 0, L_000001a7e8b82880;  alias, 1 drivers
L_000001a7e8b827e0 .cmp/gt 6, v000001a7e8b860a0_0, v000001a7e8b86140_0;
L_000001a7e8b82880 .concat [ 1 5 0 0], L_000001a7e8b827e0, L_000001a7e8bc7838;
```

steveicarus, 2021-06-13:
> statement. The .concat statement is a functor node that takes at input

Please ask questions like this in the iverilog-devel mailing list, instead
of on github.

slivebullet, 2021-06-13:
Sorry, does the iverilog-devel mailing list you mentioned refer to the SourceForge community? I don't quite understand

closed 2021-06-13 by slivebullet