steveicarus/iverilog #524: Casting to signed bus is not supported.

bitglitcher, 2021-06-15:
Hello, I'm trying to Simulate some code that is synthesizeable. But unfortunately `iverilog` gives me an error.

```
memory_access_tb.sv:158: syntax error
memory_access_tb.sv:158: error: malformed statement
```

Basically is just casting a bus into a signed bus. This works on modelsim and Quartus. But not on `iverilog`.
```verilog
32'(signed'(memory[address[2:0]]))
```
I'm using the `-g2012` flag and my current version of  `iverilog` is: `icarus Verilog version 12.0 (devel) (s20150603-1130-g1f8876be)`

bitglitcher, 2021-06-15:
Are there any alternatives that can be synthesized?

martinwhitaker, 2021-06-15:
Use $signed instead of signed'().
Closing as a duplicate of issue #513.

closed 2021-06-15 by martinwhitaker