steveicarus/iverilog #525: Use -s option without .v file

xiuwen8686, 2021-06-16:
There was an error when I compile my project before.
[image]
If I add "-s glbl" in my command,this error will solved.

I didn't add the "(vivado_addr)/data/verilog/src/glbl.v" in my command,but it still  work.
Isn't this option used to Specify a file as the topmodule? 

Thank you!

caryr, 2021-06-21:
Can you provide a small code example and the exact command line? This will be much easier to debug with actual code.

martinwhitaker, 2021-07-31:
No response so closing. If you can provide more information that would let us understand the issue, please reopen.

closed 2021-07-31 by martinwhitaker