steveicarus/iverilog #526: Excess memory usage

nrllhclb, 2021-06-20:
Hello everyone,

When I compile my Verilog a code that performs division and its testbench, I get some errors, and the process is killed. I check the system monitor, and I realized that Icarus Verilog used 4.5+GByte Ram memory, and then it was killed.

Is there any solution to this issue? As for the configuration of used memory, how do I limit the usage of memory in Icarus Verilog?

caryr, 2021-06-21:
Use a machine with more memory :)! Without some kind of code to determine if there is excessive memory usage there is not much we can do. Are you allocating large memories or other structures?

martinwhitaker, 2021-07-31:
No response, so closing. If you can provide more information that would let us reproduce the issue, please reopen.

closed 2021-07-31 by martinwhitaker