steveicarus/iverilog #528: inaccurate unnamed genblk numbering

zachjs, 2021-06-30:
In a module such as the following, the standard would appear to imply that all four blocks should be named `genblk1`. Indeed, this is the behavior I've run into in most simulators.
```Verilog
module top;
         if (P == 1) integer w = 1;
    else if (P == 2) integer x = 2;
    else if (P == 3) integer y = 3;
    else             integer z = 9;
endmodule
```

For the equivalent case statement below, all 4 blocks receive the same implicit name, as expected.
```Verilog
    case (P)
        1      : integer w = 1;
        2      : integer x = 2;
        3      : integer y = 3;
        default: integer z = 9;
    endcase
```

Further, the numbering does not reset as required in nested scopes. In the following snippet, the two nested blocks should be `genblk1` and `genblk2`, respectively, but instead continue the numbering of the outer scope.
```Verilog
    if (1) begin
        wire B = 1;
        if (1) begin
            wire C = 1;
        end
        if (1) begin
            wire D = 1;
        end
    end
```

martinwhitaker, 2021-07-27:
For your first example, my reading of the standard is different. You have three conditional generate constructs. These are directly nested, so the standard allows the generate blocks to all be given the same name. However, the "External names for unnamed generate blocks" section of the standard states:

> Each generate construct in a given scope is assigned a number. The number will be 1 for the construct that
> appears textually first in that scope and will increase by 1 for each subsequent generate construct in that
> scope.

Nowhere does it say that the directly nested constructs should be treated as a single construct. However, if all other simulators do this, we should maybe go with the herd.

I agree that the failure to reset the numbering in nested scopes is a bug.

martinwhitaker, 2021-08-04:
This exposed several more bugs. Fixed now in both the master and v11 branches.

closed 2021-08-04 by martinwhitaker