steveicarus/iverilog #529: Unexpected syntax error on macro call with concatenation

taichi-ishitani, 2021-07-04:
Hi,

I got an unexpected syntax error when parsing a macro call of which argument is concatenation like below.

```verilog
module test;
`define foo(v) \
$display("%h", v);
initial `foo({8'hab, 8'hcd})
endmodule
```

You will see the following error when you try to compile the above code.

```
$ iverilog test.v
test.v:4: error: too many arguments for `foo
test.v:5: syntax error
I give up.
```

iverilog version: 

```
$ iverilog -v
Icarus Verilog version 10.3 (stable) ()
```

martinwhitaker, 2021-07-04:
This has already been fixed (see issue #309). The current stable release (11.0) includes the fix.

closed 2021-07-04 by martinwhitaker

taichi-ishitani, 2021-07-05:
@martinwhitaker ,
Thanks for your comment! I understood.