steveicarus/iverilog #531: segfault constant evaluating function with OOB accesses

zachjs, 2021-07-27:
`iverilog` (default options) crashes when run on the following module which forces the constant evaluation of a function with out of bounds accesses.
```Verilog
module top;
    function automatic [2:0] f;
        input [2:0] i;
        begin
            f[0] = i[0];
            f[3] = i[3];
            f[3:2] = i[3:2];
        end
    endfunction
    localparam C = f(3'b110);
    initial $display("C %b", C);
endmodule
```
This issues exists both in v11.0 and on the latest master (6b127432da3ac848435200587a907738f6c3875a). Other simulators raise warnings, but simulate just fine and do not crash.

martinwhitaker, 2021-07-27:
I see an assertion failure, not a segfault:
````
ivltests/br_gh531.v:10: assert: ../source/net_func_eval.cc:369: failed assertion base + lval->lwidth() <= old_lval->expr_width()
````
(which is still a bug)

zachjs, 2021-07-27:
You are correct! I have updated the title to reflect this.

martinwhitaker, 2021-07-30:
Fixed in both the master and v11 branches.

closed 2021-07-30 by martinwhitaker