steveicarus/iverilog #532: inaccurate sizing of genvars

zachjs, 2021-07-27:
When simulating the following, iverilog produces `01100100 8`, while all other simulators produce `00000000000000000000000001100100 32`.
```SystemVerilog
module top;
    genvar i;
    for (i = 100; i < 101; i = i + 1)
        initial $display("%b %0d", i, $bits(i));
endmodule
```

This issue exists both in v11.0 and the latest master.

From Section 27.4:
> Within the generate block of a loop generate construct, there is an implicit localparam declaration. This is an integer parameter that has the same name and type as the loop index variable, and its value within each instance of the generate block is the value of the index variable at the time the instance was elaborated.

While I'm a little thrown off by the "and type", the majority interpretation appears to be that genvars should simply be sized as integers.

martinwhitaker, 2021-07-28:
You can get the correct behaviour by using the `-gstrict-expr-width` compiler option.

zachjs, 2021-07-29:
I wasn't aware of this option. Thank you!

closed 2021-07-29 by zachjs