steveicarus/iverilog #535: incomplete detection of number literal truncation with -gstrict-expr-width

zachjs, 2021-08-01:
For each of the number literals, I am using the following stub:
```Verilog
module top;
    localparam X = <number_literal_here>;
    initial $display("%0d %0d %b", $bits(X), X, X);
endmodule
```

For the number literal `4294967294`, I get `33 4294967294 011111111111111111111111111111110` without the flag, and `32 -2 11111111111111111111111111111110` with the flag, but no warnings are produced in the latter mode, despite truncation occurring. This appears to be due to `truncate_to_integer_width` tolerating 0 padding for such literals, despite the automatically added leading zero bit not being optional for the unsized signed decimals.

Interestingly, iverilog does not, and apparently should not (based on commercial simulators), warn for the implicit zero-pad clipping occurring in literals like `32'sd4294967294` and `1'sd1`.

For the number literal `'bZ_01XZ01XZ_01XZ01XZ_01XZ01XZ_01XZ01XZ`, I get `33 X z01xz01xz01xz01xz01xz01xz01xz01xz` without the flag, and `32 X 01xz01xz01xz01xz01xz01xz01xz01xz` with the flag, but no warnings are produced in the latter mode, despite truncation occurring. If I change the most significant bit to a 1, I correctly get a warning with `-gstrict-expr-width`. If I instead explicitly size the literal to 32 bits, I correctly get a truncation warning in both modes. This appears to be due to `truncate_to_integer_width` basing the allowed padding off of the most significant bit, rather than the 31st (0-indexed) bit.

zachjs, 2021-08-01:
I've been poking around some more, and I'm not entirely convinced something like `'hF_FFFF_FFFF` should be truncated or produce a warning in either mode.

> The number of bits that make up an unsized number (which is a simple decimal number or a number with a base specifier but no size specification) shall be at least 32. Unsized unsigned literal constants where the high-order bit is unknown (X or x) or three-state (Z or z) shall be extended to the size of the expression containing the literal constant.

Obviously this whole section of the LRM is underspecified, but I'm curious what has informed your decision regarding truncation of unsized literals in the stricter mode.

martinwhitaker, 2021-08-01:
The LRM is indeed unclear. See https://www.accellera.org/images/eda/sv-bc/11350.html for an explanation of the actual intent. An unsized literal is meant to be the same size as an integer. In IEEE 1364 (standard Verilog), that is at least 32 bits. In IEEE 1800 (SystemVerilog) it is exactly 32 bits. Icarus Verilog defaults to 32 bits, but has a configuration option that lets you change it. I don't know of any other simulator that supports anything other than 32 bits.

As far as truncation is concerned, from IEEE 1800-2012 section 5.7.1

> If the size of the unsigned number is larger than the size specified for the literal constant, the unsigned number shall be truncated from the left.

so as an unsized literal constant has an implicit size of 32 bits, we should truncate it if the value exceeds 32 bits.

zachjs, 2021-08-01:
Thank you very much for sharing that thread! This has been very helpful.

Now, my original issue regarding warnings which are not issued but should be is valid, I believe.

While we're on the topic of these truncation warnings, I've noticed some further interesting results.

| Literal | iverlog | iverilog -gstrict-expr-width | xcelium |
| - | - | - | - |
| `'hZ_ZZZZ_ZZZZ` | no warning | truncates, no warning | truncates, warns |
| `'hF_FFFF_FFFF ` | no warning | truncates, warns | truncates, warns |
| `'h0_FFFF_FFFF ` | no warning | truncates, no warning | truncates, no warning |
| `'sh0_FFFF_FFFF ` | no warning | truncates, no warning | truncates, no warning |
| `32'hZ_ZZZZ_ZZZZ`  | truncates, warns | truncates, warns | truncates, warns |
| `32'hF_FFFF_FFFF ` | truncates, warns | truncates, warns | truncates, warns |
| `32'h0_FFFF_FFFF ` | truncates, warns | truncates, warns | truncates, no warning |
| `32'sh0_FFFF_FFFF ` | truncates, warns | truncates, warns | truncates, no warning |

I'd argue that iverilog's behavior is reasonable here, except that with `-gstrict-expr-width`, a warning should be produced in all cases. What do you think?

martinwhitaker, 2021-08-04:
There are two types of warnings involved - excess digits and value truncation. I tested four other simulators. All but one warned about excess digits for the sized values. None warned about excess digits for the unsized values. Apart from iverilog, only xcelium warns about value truncation.

The argument for not warning about the `'hz_zzzz_zzzz` truncation is that, from the LRM:

> Unsized unsigned constants where the high-order bit is unknown ( X or x ) or three-state ( Z or z ) shall be extended to the size of the expression containing the constant.

So although it is truncated, you get the same value back when you use it.

The argument for not warning about the `'h0_ffff_ffff` truncation is that it still represents the same value. And, strictly speaking, that is true for `'sh0_ffff_ffff` as well, because the value after the `'sh` is unsigned. The fact that we match the xcelium behaviour here seems a good argument for keeping the status quo.

zachjs, 2021-08-06:
Thanks for your explanation! The reasoning for `'hz_zzzz_zzzz` makes sense to me. I do still think the lack of a warning for `'sh0_ffff_ffff` is a little strange, as the truncation changes the value from 2^32 to -1, changing how it sign extends in a surprising (and silent!) way.

In any case, I think the lack of warnings for the original two test cases (being `4294967294` and `'bZ_01XZ01XZ_01XZ01XZ_01XZ01XZ_01XZ01XZ`) is still unintended. Is this true?

martinwhitaker, 2021-08-15:
My intention was to emit the same value truncation warnings for unsized literals as for 32-bit sized literals (but no excess digit warnings), and I believe it does that. That's not to say the warnings are perfect in either case.

martinwhitaker, 2024-01-08:
I'm going to label this as an enhancement, in case someone else wants to look at improving/extending the compiler warnings.