steveicarus/iverilog #536: support unpacked assignments

zachjs, 2021-08-03:
I believe the following is valid in both Verilog-2005 and SystemVerilog-2017. iverilog currently complains: `error: Cannot assign to array arr1['sd0]. Did you forget a word index?.
```Verilog
module top;
    reg [7:0] arr1 [0:7][0:1];
    reg [7:0] arr2 [0:7][0:1];
    initial arr1[0] = arr2[0];
endmodule
```

martinwhitaker, 2021-08-04:
Reading or writing a slice of an unpacked array is a SystemVerilog feature, which is currently not supported.

ikle, 2024-11-02:
As I understand it, this is the same, or almost the same, as passing an array slice as a parameter when instantiating a module. At the moment I get messages like:

    redux.v:38: sorry: Array slices are not yet supported for continuous assignment.
    redux.v:38:      : Port 1 (x) of redux is connected to x[(i)*('sd3):((i)*('sd3))+('sd2)]
    redux.v:38: sorry: Array slices are not yet supported for continuous assignment.
    redux.v:38:      : Port 2 (q) of redux is connected to y[(i)*('sd2):((i)*('sd2))+('sd1)]