steveicarus/iverilog #537: documentation for iVerilog pli/vpi Integration in C/C++ applications

carlosdelfino, 2021-08-06:
Hello.
  I'm studying the development of a QT application where I integrate iVerilog with a simulated hardware maintenance and diagnostics interface, my goal is to have greater control of my project and visualize internal data in registers and bram.

I would like to know if there is any documentation I can read about integrating iVerilog into applications, beyond what is available at: https://iverilog.fandom.com/wiki/Using_VPI

Grateful.

carlosdelfino, 2021-08-09:
I'm having success using QTcreator to edit and compile code for iVerilog, it's not problematic, it's pretty simple, but I can't link with already compiled libraries from iVerilog 10 which is distributed by apt-get special on Ubuntu 18.04 (i386) .

I even thought about compiling the iVerilog version 11 source, but wanted to solve the problem using a standard installation per hour.

The error I get when trying to link is as follows:

```
simulinho_vpi.o: In function `hello_register()':
/home/carlosdelfino/workspace/RISCuinho/SIMULinho-build-Desktop_RISCuinho-Profile/../SIMULinho/simulinho_vpi.cpp:25: undefined reference to `vpi_register_systf'
simulinho_vpi.o: In function `hello_calltf(char*)':
/home/carlosdelfino/workspace/RISCuinho/SIMULinho-build-Desktop_RISCuinho-Profile/../SIMULinho/simulinho_vpi.cpp:11: undefined reference to `vpi_printf'
collect2: error: ld returned 1 exit status
Makefile:157: recipe for target 'SIMULinho' failed
make: *** [SIMULinho] Error 1
``` 

Below are the commands and their parameters used to generate the object file and the one used to link with my application:

```
g++ -pipe -O2 -g -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DSIMULINHO_SHAREDLIB -D__SIMULINHO_DUMP__ -DQT_QML_DEBUG -DQT_NO_DEBUG -DQT_QUICK_LIB -DQT_GUI_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -fstack-protector-strong -Wformat -Werror=format-security -Wall -Wextra -Wshadow -fdebug-prefix-map=/build/iverilog-3pPO9t/iverilog-10.1=. -fPIC -I../SIMULinho -I. -I../SIMULinho -isystem /usr/include/iverilog -isystem /usr/include/i386-linux-gnu/qt5 -isystem /usr/include/i386-linux-gnu/qt5/QtQuick -isystem /usr/include/i386-linux-gnu/qt5/QtGui -isystem /usr/include/i386-linux-gnu/qt5/QtQml -isystem /usr/include/i386-linux-gnu/qt5/QtNetwork -isystem /usr/include/i386-linux-gnu/qt5/QtCore -I. -isystem /usr/include/libdrm -I/usr/lib/i386-linux-gnu/qt5/mkspecs/linux-g++ -c ../SIMULinho/simulinho_vpi.cpp -o simulinho_vpi.o

g++ -c -pipe -O2 -g -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DSIMULINHO_SHAREDLIB -D__SIMULINHO_DUMP__ -DQT_QML_DEBUG -DQT_NO_DEBUG -DQT_QUICK_LIB -DQT_GUI_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../SIMULinho -I. -I../SIMULinho -isystem /usr/include/iverilog -isystem /usr/include/i386-linux-gnu/qt5 -isystem /usr/include/i386-linux-gnu/qt5/QtQuick -isystem /usr/include/i386-linux-gnu/qt5/QtGui -isystem /usr/include/i386-linux-gnu/qt5/QtQml -isystem /usr/include/i386-linux-gnu/qt5/QtNetwork -isystem /usr/include/i386-linux-gnu/qt5/QtCore -I. -isystem /usr/include/libdrm -I/usr/lib/i386-linux-gnu/qt5/mkspecs/linux-g++ -o main.o ../SIMULinho/main.cpp
g++ -c -pipe -O2 -g -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DSIMULINHO_SHAREDLIB -D__SIMULINHO_DUMP__ -DQT_QML_DEBUG -DQT_NO_DEBUG -DQT_QUICK_LIB -DQT_GUI_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../SIMULinho -I. -I../SIMULinho -isystem /usr/include/iverilog -isystem /usr/include/i386-linux-gnu/qt5 -isystem /usr/include/i386-linux-gnu/qt5/QtQuick -isystem /usr/include/i386-linux-gnu/qt5/QtGui -isystem /usr/include/i386-linux-gnu/qt5/QtQml -isystem /usr/include/i386-linux-gnu/qt5/QtNetwork -isystem /usr/include/i386-linux-gnu/qt5/QtCore -I. -isystem /usr/include/libdrm -I/usr/lib/i386-linux-gnu/qt5/mkspecs/linux-g++ -o simulinho.o ../SIMULinho/simulinho.cpp
/usr/lib/qt5/bin/rcc -name qml ../SIMULinho/qml.qrc -o qrc_qml.cpp
g++ -c -pipe -O2 -g -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DSIMULINHO_SHAREDLIB -D__SIMULINHO_DUMP__ -DQT_QML_DEBUG -DQT_NO_DEBUG -DQT_QUICK_LIB -DQT_GUI_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../SIMULinho -I. -I../SIMULinho -isystem /usr/include/iverilog -isystem /usr/include/i386-linux-gnu/qt5 -isystem /usr/include/i386-linux-gnu/qt5/QtQuick -isystem /usr/include/i386-linux-gnu/qt5/QtGui -isystem /usr/include/i386-linux-gnu/qt5/QtQml -isystem /usr/include/i386-linux-gnu/qt5/QtNetwork -isystem /usr/include/i386-linux-gnu/qt5/QtCore -I. -isystem /usr/include/libdrm -I/usr/lib/i386-linux-gnu/qt5/mkspecs/linux-g++ -o qrc_qml.o qrc_qml.cpp
```

And finally the final command that is giving an error for linking objects:
```
g++  -o SIMULinho simulinho_vpi.o main.o simulinho.o qrc_qml.o   -lvpi -lveriuser -L/usr/lib/i386-linux-gnu -lQt5Quick -lQt5Gui -lQt5Qml -lQt5Network -lQt5Core -lGL -lpthread 
```

caryr, 2021-08-14:
The wiki and the IEEE 1800 standard are the best place to get information. Cocotb also uses the VPI to access simulator state information. Icarus does not fully support the VPI, but it should support enough to do what you want to do.

carlosdelfino, 2021-08-15:
Generating .vpi files (shared objects) integrated with QT.

I was successful in compiling, however when I try to integrate the generated `.vpi` file, I get an error message indicating that some references are missing.

I don't know if the problem is in my compilation, or in fact I need to pass some parameter to vvp when loading my functions/tasks module defined in the `.vpi` file.

I even opened a question [StackOverflow](https://stackoverflow.com/questions/68791604/using-qt5-and-qtquick2-objects-and-libraries-in-custom-shared-libraries)

The error message I get is below:

```
iverilog -o simulinho.vvp /home/carlosdelfino/workspace/RISCuinho/SIMULinho/simulinho.v && vvp -M. -msimulinho_vpi simulinho.vvp
simulinho_vpi:`./simulinho_vpi.vpi' failed to open using dlopen() because:
    ./simulinho_vpi.vpi: undefined symbol: _Z21qRegisterResourceDataiPKhS0_S0_.
/home/carlosdelfino/workspace/RISCuinho/SIMULinho/simulinho.v:3: Error: System task/function $hello() is not defined by any module.
/home/carlosdelfino/workspace/RISCuinho/SIMULinho/simulinho.v:4: Error: System task/function $openUI() is not defined by any module.
/home/carlosdelfino/workspace/RISCuinho/SIMULinho/simulinho.v:5: Error: System task/function $closeUI() is not defined by any module.
simulinho.vvp: Program not runnable, 3 errors.
Makefile:358: recipe for target 'verilognize_vvp' failed
make: *** [verilognize_vvp] Error 3
``` 
Should I pass some more information to vvp to be able to correctly use this module?

martinwhitaker, 2021-08-15:
What library provides the _Z21qRegisterResourceDataiPKhS0_S0_ symbol. Did you include it in the list of libraries to be linked when building your .vpi file?

carlosdelfino, 2021-08-15:
I think is QTCore, yes I put it:

``` 
 g++ --shared -pipe -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_VERSION_TAGGING -DSIMULINHO_SHAREDLIB -D__SIMULINHO_DUMP__ -DQT_NO_DEBUG -DQT_QUICK_LIB -DQT_GUI_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB  -L/usr/lib/i386-linux-gnu -lvpi -lveriuser -lQt5Quick -lQt5Gui -lQt5Qml -lQt5Network -lQt5Core -lGL -lpthread  simulinho_vpi.o qrc_qml.o -o simulinho_vpi.so simulinho.o
```

martinwhitaker, 2021-08-16:
I created this simple C++ example (qtapp.cc)
````
#include <QApplication>
#include <QWidget>

#include <vpi_user.h>

static PLI_INT32 qtapp_calltf(PLI_BYTE8 *xx)
{
    (void)xx;  // parameter is not used
    
    int argc = 1;
    char *argv[] = { (char *)"QTAPP" };
 
    QApplication app(argc, argv); 
    
    QWidget window;
    
    window.resize(250, 150);
    window.setWindowTitle("QTAPP GUI");
    window.show();
 
    vpi_printf("Starting GUI\n");
    
    return app.exec();
}   

static void qtapp_register(void)
{
    s_vpi_systf_data tf_data;
    
    tf_data.type      = vpiSysTask;
    tf_data.tfname    = "$qtapp";
    tf_data.calltf    = qtapp_calltf;
    tf_data.compiletf = 0;
    tf_data.sizetf    = 0;
    vpi_register_systf(&tf_data);
}

void (*vlog_startup_routines[])(void) = {
    qtapp_register,
    0
};  
````
and compiled it with
````
iverilog-vpi -I/usr/include/qt5 -I/usr/include/qt5/QtWidgets -lQt5Widgets -lQt5Core qtapp.cc
````
You may need to adjust the search paths for your system - you can add a -L option if necessary.

Then I created this Verilog file (qtapp.v)
````
module main();

initial begin
    $qtapp;
    $finish(0);
end
 
endmodule
````
compiled it with
````
iverilog -L . -m qtapp -o qtapp qtapp.v
````
and ran it with
````
vvp qtapp
````
This ran without error, displaying the "QTAPP GUI" window,

carlosdelfino, 2021-08-16:
To compile an object that uses Qt5 it takes a lot of parameters, and it seems that not all of them are accepted by iverilog-vpi.

```
-std=gnu++11 
-Wall -W 

-fPIC 

-D_REENTRANT -DQT_DEPRECATED_WARNINGS -DQT_NO_VERSION_TAGGING -DQT_QUICK_LIB -DQT_GUI_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB

-I. -I/usr/local/lib/ivl/include 

-isystem /usr/local/include/iverilog -isystem /usr/include/i386-linux-gnu/qt5 -isystem /usr/include/i386-linux-gnu/qt5/QtQuick -isystem /usr/include/i386-linux-gnu/qt5/QtGui -isystem /usr/include/i386-linux-gnu/qt5/QtQml -isystem /usr/include/i386-linux-gnu/qt5/QtNetwork -isystem /usr/include/i386-linux-gnu/qt5/QtCore 

-isystem /usr/include/libdrm -I/usr/lib/i386-linux-gnu/qt5/mkspecs/linux-g++
```

carlosdelfino, 2021-08-17:
It was just the order of the parameters, now everything worked:

```
 make verilognize
/usr/lib/qt5/bin/qmake -o Makefile ../SIMULinho.old/SIMULinho.pro -spec linux-g++ CONFIG+=debug CONFIG+=qml_debug
rm -f simulinho_vpi.o simulinho_vpi.so
rm -f simulinho_vpi.vpi simulinho_vpi.vpi.debug
g++ -c -pipe -g -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_VERSION_TAGGING -DSIMULINHO_SHAREDLIB -D__SIMULINHO_DUMP__ -DQT_QUICK_LIB -DQT_GUI_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -fPIC -I../SIMULinho.old -I. -I/usr/local/lib/ivl/include -isystem /usr/local/include/iverilog -isystem /usr/include/i386-linux-gnu/qt5 -isystem /usr/include/i386-linux-gnu/qt5/QtQuick -isystem /usr/include/i386-linux-gnu/qt5/QtGui -isystem /usr/include/i386-linux-gnu/qt5/QtQml -isystem /usr/include/i386-linux-gnu/qt5/QtNetwork -isystem /usr/include/i386-linux-gnu/qt5/QtCore -I. -isystem /usr/include/libdrm -I/usr/lib/i386-linux-gnu/qt5/mkspecs/linux-g++ -I/usr/include/i386-linux-gnu/qt5/QtGui -I/usr/include/i386-linux-gnu/qt5/QtWidgets -fstack-protector-strong -Wformat -Werror=format-security -Wextra -Wshadow -fdebug-prefix-map=/build/iverilog-3pPO9t/iverilog-10.1=. -g -o simulinho_vpi.o ../SIMULinho.old/simulinho_vpi.cpp && g++ -o simulinho_vpi.so --shared -pipe -g -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_VERSION_TAGGING -DSIMULINHO_SHAREDLIB -D__SIMULINHO_DUMP__ -DQT_QUICK_LIB -DQT_GUI_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB simulinho_vpi.o qrc_qml.o simulinho.o  -L/usr/lib/i386-linux-gnu -L/usr/local/lib/ -lvpi -lveriuser -lQt5Quick -lQt5Gui -lQt5Qml -lQt5Network -lQt5Core -lGL -lpthread 
../SIMULinho.old/simulinho_vpi.cpp: In function ‘int simulinho_ui_compiletf(char*)’:
../SIMULinho.old/simulinho_vpi.cpp:13:40: warning: unused parameter ‘user_data’ [-Wunused-parameter]
 static int simulinho_ui_compiletf(char*user_data){
                                        ^~~~~~~~~
../SIMULinho.old/simulinho_vpi.cpp: In function ‘int simulinho_ui_calltf(char*)’:
../SIMULinho.old/simulinho_vpi.cpp:17:37: warning: unused parameter ‘user_data’ [-Wunused-parameter]
 static int simulinho_ui_calltf(char*user_data){
                                     ^~~~~~~~~
../SIMULinho.old/simulinho_vpi.cpp: In function ‘int hello_compiletf(char*)’:
../SIMULinho.old/simulinho_vpi.cpp:27:33: warning: unused parameter ‘user_data’ [-Wunused-parameter]
 static int hello_compiletf(char*user_data)
                                 ^~~~~~~~~
../SIMULinho.old/simulinho_vpi.cpp: In function ‘int hello_calltf(char*)’:
../SIMULinho.old/simulinho_vpi.cpp:32:30: warning: unused parameter ‘user_data’ [-Wunused-parameter]
 static int hello_calltf(char*user_data)
                              ^~~~~~~~~
cp --remove-destination simulinho_vpi.so simulinho_vpi.vpi
iverilog -L . -L /usr/include/i386-linux-gnu/qt5/QtGui -L /usr/include/i386-linux-gnu/qt5/QtWidgets -m simulinho_vpi -o simulinho.vvp /home/carlosdelfino/workspace/RISCuinho/SIMULinho.old/simulinho.v && vvp -v simulinho.vvp
Hell Register!Simulinho UI Open Register!Simulinho UI Close Register!Compiling VVP ...
 ... VVP file version 11.0 (stable) (v11_0-72-gd3b0992a)
Hell Register!Simulinho UI Open Register!Simulinho UI Close Register!Compile cleanup...
 ... Linking
 ... Removing symbol tables
 ... Compiletf functions
 ...        0 functors (net_fun pool=524288 bytes)
                  0 logic
                  0 bufif
                  0 resolv
                  0 signals
 ...        0 filters (net_fil pool=524288 bytes)
 ...        7 opcodes (16384 bytes)
 ...        0 nets
 ...        0 vvp_nets (0 bytes)
 ...        0 arrays (0 words)
 ...        0 memories
                  0 logic (0 words)
                  0 real (0 words)
 ...        1 scopes
 ... 0.015737 seconds, 67996.0/11308.0/9656.0 KBytes size/rss/shared
Running ...
 ...execute EndOfCompile callbacks
 ...propagate initialization events
 ...execute StartOfSim callbacks
 ...run scheduler
Hello, World!
Executando com Argumentos: SIMULINHO VPIExecutando com Argumentos: SIMULINHO VPI ...execute Postsim callbacks
 ... 0,706909 seconds, 150316,0/45488,0/32884,0 KBytes size/rss/shared
Event counts:
           4 time steps (pool=204)
           4 thread schedule events
           0 assign events
             ...assign(vec4) pool=16384
             ...assign(vec8) pool=409
             ...assign(real) pool=409
             ...assign(word) pool=256
             ...assign(word/r) pool=341
           0 other events (pool=8192)
echo V E R I L O G N I Z E
V E R I L O G N I Z E
```

martinwhitaker, 2021-08-17:
Good to hear you've found the fault. Closing the issue.

closed 2021-08-17 by martinwhitaker