steveicarus/iverilog #538: $system task not supported by VPP

dineshannayya, 2021-08-16:
I see tool not able to understand the $system command and issue error as

Error: System task/function $system() is not defined by any module

How to invoke $system call from verilog ?

caryr, 2021-08-16:
Yes, this is currently no implemented. It should be easy to add. I have purposely not added it so far because using $system() opens SystemVerilog up to security issues (e.g. arbitrary shell code can be executed). My though was to require the user to add a flag acknowledging they understand the security risk and wish to enable this functionality. Of course I would also like to see if @steveicarus  and @martinwhitaker  agree.

martinwhitaker, 2021-08-17:
@caryr, yes, this could be a security hole, so not enabling it by default seems prudent.

steveicarus, 2021-08-19:
Or at all. What's the use case?

dineshannayya, 2021-08-19:
I have a Riscv based test environment, which needs system tasks for test
bench automation.  Currently I have created a system task using VPI and
it's working fine.

chykon, 2023-09-17:
`$system` can be useful for performing a trick like this (https://stackoverflow.com/a/66855217) related to getting a random seed value for the RNG.