steveicarus/iverilog #539: `always_comb` block unable to extract sensitivity list

AmeyaVS, 2021-08-21:
I am trying to simulate the following generated SV module along with it's testbench:

**Design: `counter.sv`**

```systemverilog
//==============================================================================
//
// The code is generated by Intel Compiler for SystemC, version 1.3.31
// see more information at https://github.com/intel/systemc-compiler
//
//==============================================================================

//==============================================================================
//
// Module: Dut ()
//
module Dut // "tb.dut_inst"
(
    input logic clk,
    input logic rstn,
    output logic [3:0] counter,
    output logic even
);

//------------------------------------------------------------------------------
// Clocked THREAD: threadProc (dut.h:26:5) 

// Thread-local variables
logic [3:0] counter_next;

// Next-state combinational logic
always_comb begin : threadProc_comb     // dut.h:26:5
    threadProc_func;               //<   Comment this line for iverilog simulation to work
    //counter_next = counter;      //< Uncomment this line for iverilog simulation to work
    //counter_next = counter + 1;  //< Uncomment this line for iverilog simulation to work
end
function void threadProc_func;
    counter_next = counter;
    counter_next = counter + 1;
endfunction

// Syncrhonous register update
always_ff @(posedge clk or negedge rstn) 
begin : threadProc_ff
    if ( ~rstn ) begin
        counter <= 0;
    end
    else begin
        counter <= counter_next;
    end
end

//------------------------------------------------------------------------------
// Method process: methodProc (dut.h:35:5) 

always_comb 
begin : methodProc     // dut.h:35:5
 even = |(counter % 2);
end

endmodule
```
**Design Testbench: `tb_counter.sv`**
```systemverilog
module tb_counter();

logic clk;
logic rstn;
logic [3:0] counter;
logic even;

Dut DUT0(clk, rstn, counter, even);

always
  #10  clk= ~clk;

initial
begin
    rstn = 1;
    clk = 0;
    #20 rstn = 0;

    #30 rstn = 1;

    #500 $finish;
end

initial begin
        if ($test$plusargs("trace") != 0) begin
                $display("[%0t] Tracing to vlt_dump.vcd...\n", $time);
                $dumpfile("logs/vlt_dump.vcd");
                $dumpvars;
        end
        $display("[%0t] Model Running...\n", $time);
end

endmodule
```
Getting the following error while trying out build and run the simulation:

**Iverilog Build command:**
```bash
iverilog -g2012 counter.sv tb_counter.sv -o counter.aout
```
**Output:**
```
counter.sv:27: warning: always_comb process has no sensitivities.
ivl: ../net_scope.cc:500: const NetTaskDef* NetScope::task_def() const: Assertion `type_ == TASK' failed.
```

If one comments the line `28` in `counter.sv` and un-comment the lines `29-30` iverilog is able to extract sensitivity.

Here are the details below for the `iverilog` build from source:

**Iverilog Build info:`iverilog -v`**

```txt
iverilog -v
Icarus Verilog version 12.0 (devel) (s20150603-1147-g7ee7a483)

Copyright (c) 2000-2021 Stephen Williams (steve@icarus.com)

  This program is free software; you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation; either version 2 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License along
  with this program; if not, write to the Free Software Foundation, Inc.,
  51 Franklin Street, Fifth Floor, Boston, MA 02110-1301, USA.

iverilog: no source files.

Usage: iverilog [-EiSuvV] [-B base] [-c cmdfile|-f cmdfile]
                [-g1995|-g2001|-g2005|-g2005-sv|-g2009|-g2012] [-g<feature>]
                [-D macro[=defn]] [-I includedir] [-L moduledir]
                [-M [mode=]depfile] [-m module]
                [-N file] [-o filename] [-p flag=value]
                [-s topmodule] [-t target] [-T min|typ|max]
                [-W class] [-y dir] [-Y suf] [-l file] source_file(s)

See the man page for details.
```
Let me know in-case any more details are required.

svenka3, 2021-08-21:
Looks like Icarus uses always@(*) semantics to extract sensitivity list and
does not support the inference via a function (inside always comb).

Regards
Srini

martinwhitaker, 2021-08-22:
@svenka3, no, that's incorrect. It handles proper functions correctly. The problem here is that the compiler treats a void function as a task, and task calls do not contribute to the sensitivity list.

svenka3, 2021-08-22:
Thanks for the clarification. So if it was a non-void function, it should
work. Maybe there is a workaround in that Intel SystemC compiler from OP?

Regards
Srini

caryr, 2021-09-28:
The work around is to detect void functions correctly and then treat them like a normal functions when calculating the sensitivity list.

PR #831 (2022-12-27): Let void functions in always_comb block contribute to sensitivity list
Internally void function calls are modeled as task calls. But unlike task
calls, access to a signal inside a void function is supposed to contribute
to the sensitivity list of an always_comb block.

Make sure that the same logic applied for non-void functions is also
applied for void functions.

Resolves #539
closed 2022-12-27 by steveicarus