steveicarus/iverilog #540: inaccurate combination if signedness of non-ANSI style port declarations

zachjs, 2021-09-05:
Per 23.2.2.1 of IEEE 1800-2017:

> The signed attribute can be attached either to a port declaration or the corresponding net or variable declaration or to both. If either the port or the net/variable is declared as signed, then the other shall also be considered signed.

Contrary to this specification, iverilog appears to only consider the signedness of the data declaration, rather considering the port signed if either declaration marks it as such.

```SystemVerilog
module top(
    o___one, o___two,
    o__signed_one, o__signed_two,
    o__unsigned_one, o__unsigned_two,
    o_signed__one, o_signed__two,
    o_signed_signed_one, o_signed_signed_two,
    o_signed_unsigned_one, o_signed_unsigned_two,
    o_unsigned__one, o_unsigned__two,
    o_unsigned_signed_one, o_unsigned_signed_two,
    o_unsigned_unsigned_one, o_unsigned_unsigned_two
);

`define TEST_RAW(name, port, data) \
    output port name; \
    wire data name; \
    assign name = 1'sb1; \
    wire [$bits(name):0] tmp_``name = name; \
    initial #1 $display(`"name %b %b`", name, tmp_``name);

`define TEST_SG(port_sg, data_sg) \
    `TEST_RAW(o_``port_sg``_``data_sg``_one, port_sg      , data_sg      ) \
    `TEST_RAW(o_``port_sg``_``data_sg``_two, port_sg [1:0], data_sg [1:0]) \

    `TEST_SG(        ,         )
    `TEST_SG(        ,   signed)
    `TEST_SG(        , unsigned)
    `TEST_SG(  signed,         )
    `TEST_SG(  signed,   signed)
    `TEST_SG(  signed, unsigned)
    `TEST_SG(unsigned,         )
    `TEST_SG(unsigned,   signed)
    `TEST_SG(unsigned, unsigned)
endmodule
```

martinwhitaker, 2021-09-10:
Agreed this is a bug, although it is a bit weird that the standard allows a port to be declared as signed and the corresponding variable to be declared as unsigned (or vice versa). I guess the SV standard committee just didn't think it through.

martinwhitaker, 2021-09-10:
Fixed in both the master and v11 branches.

closed 2021-09-10 by martinwhitaker