steveicarus/iverilog #545: HTTPS for homepage

mcandre, 2021-09-28:
Can we get SSL certs going for the iverilog homepage?

nasciontem, 2021-09-28:
If the website already have a valid SSL cert you can redirect the users if they try to access the insecure version of the homepage (via http) to its secure version, with SSL. You have to first check if your host has already suplied the certification to you, if not you need to order it, activate it and then insert on your website a .htaccess file to automate the redirection. 
The .htaccess file will look a lot like this:

<pre><code>
RewriteEngine On
RewriteCond %{HTTPS} off
RewriteRule (.*) https://%{HTTP_HOST}%{REQUEST_URI}
</code></pre>

steveicarus, 2022-05-21:
The page hosted on github now uses https.

closed 2022-05-21 by steveicarus