steveicarus/iverilog #546: Option to terminate on indeterminate value

mcandre, 2021-09-28:
Could there be a command line flag to instruct vvp to terminate when a value enters an indeterminate state? Ideally the variable name and the source line number would print to the console.

This is helpful in terms of debugging.

For example, a very simple unit test framework implementation would be to initialize some variables with the expected values, and then proceed to run some modules. If the module tests fail, then vvp would witness this as an indeterminate state. Then the user gets a helpful trace of the offending test.

mcandre, 2021-09-28:
Well, I am learning how to use `assert` better now, so this is less of a concern.

closed 2021-09-28 by mcandre

caryr, 2021-09-28:
Also in the development branch there is very preliminary support for the elaboration system tasks. At the moment they only take a single string argument or for fatal the finish number and a string. This can be used to validate parameters at compile time using generate if constructs.