steveicarus/iverilog #547: Bad exit code on assertion failures

mcandre, 2021-09-28:
vvp continues to present a zero exit code to the shell, even though my test suite assertions fail.

Can we please update vvp to present proper `EXIT_FAILURE` codes at termination?

https://en.cppreference.com/w/cpp/utility/program/EXIT_status

caryr, 2021-09-28:
A Verilog failure is not a program failure and that is what VVP returns by default. This was discussed long ago and I added $finish_and_return(<value>); to allow the Verilog code to make vvp return a specific value. See https://iverilog.fandom.com/wiki/Verilog_Portability_Notes for the description of this non-standard task and other items.

mcandre, 2021-09-28:
Is the behavior you are describing defined in the IEEE SystemVerilog standard? Or is this just the convention so far? As a developer, I don't like how that behavior fools CI/CD into thinking that the test suite is passing.

Hmm, I appreciate the idea of a custom $finish_and_return, though I do want my code to be portable. Hmm, perhaps an optional command line flag would help.

caryr, 2021-09-29:
The standard does not cover this and I believe we match how other simulators work or at least did last time this was discussed. You do need to understand your tools before integrating them into a CI environment and given the first thing you should write is a failing test this behavior should be discovered and worked around very early.

martinwhitaker, 2021-10-15:
The portable option would be to use $fatal, e.g.
````
assert (condition) else $fatal;
````

martinwhitaker, 2024-01-08:
Closing as invalid, as there is a standard solution to get the requested behaviour.

closed 2024-01-08 by martinwhitaker