steveicarus/iverilog #548: inaccurate assert property implementation

mcandre, 2021-09-28:
assert property does not appear to emit test failures.

Would be good for vvp to warn about incomplete implementation of `assert property` until that is fixed.

Sometimes, the parser even ignores bad code surrounding assert property statements. For example, I tried `assert property (false);`, and the parser didn't report the bad literal.

caryr, 2021-09-29:
The compiler should warn about this missing functionality.

martinwhitaker, 2021-10-15:
Compiling a file containing the line
````
assert property (false);
````
results in the compiler error message
````
test.v:3: sorry: concurrent_assertion_item not supported. Try -gno-assertions or -gsupported-assertions to turn this message off.
````
Please provide a complete example that allows the issue to be reproduced.

martinwhitaker, 2022-03-26:
No response after more than 6 months, so closing.

closed 2022-03-26 by martinwhitaker