steveicarus/iverilog #549: Trouble scheduling assertions

mcandre, 2021-09-28:
Hi,

I am trying to clean up my test suite by using `assert();` well. However, there appears to be no good way to schedule assertions for scanning wire values. Assertions appear to evaluate in the middle of voltage changes, rather than as the very last part of the simulator tick, after everything else.

caryr, 2021-09-29:
Only immediate assertions are implemented in Icarus so far, what you are likely looking for are deferred assertions.

mcandre, 2021-09-29:
Are deferred assertions the same as property assertions?

caryr, 2021-09-29:
No they are different, though the `#0` version is evaluated in the observed region which I believe is the same for properties. I do not use properties in my day work, but I do use deferred assertions frequently. The other version of deferred assertion is the `final` one which is evaluated in the postponed region.

It is good to have a copy of the standard since it covers much of this in detail and since it can be downloaded for free there is really no reason not to have a copy.