steveicarus/iverilog #550: Support for always #0 syntax

mcandre, 2021-09-28:
iverilog appears to reject `always #0`... syntax. The parser mistakes `#0` as expressing zero delay, when in fact the meaning of `#0` is about pinning evaluation until the end of the current timestep.

As a workaround, I am nesting my `#0` expressions inside of a larger `#1` block:

```systemverilog
always #1 begin
    #0 ...
end
```

caryr, 2021-09-29:
Your understand of `#0` is incorrect. Icarus is correct in reporting `#0` will create an infinite loop when it is the only delay statement in an always loop. Unless your really know what you are doing `#0` is likely being used incorrectly.

mcandre, 2021-09-29:
Thank you for clarifying about `always #0` so that I will know better.

closed 2021-09-29 by mcandre