steveicarus/iverilog #551: Homebrew package

mcandre, 2021-09-29:
iverilog is proving to be super convenient for embedded projects! Can we get a Homebrew package going, in order to streamline the install process for Mac users?

https://brew.sh/

caryr, 2021-09-29:
We compile for Mac as part of our CI testing, but we primarily focus on development and leave packaging and distribution to others. Feel free to take this task if you feel qualified.

logicmonkey, 2021-12-20:
There is a homebrew package: `icarus-verilog`

Personally I don't use it because while `iverilog` has clearly built OK, it won't **run** my code. Instead I build locally on my M1 MBA and run from there. But for it to work, you need a `bison` install that is waaaaay more recent than Xcode's rev 2.3 out of the box.

And this is why I think the homebrew version is broken - it uses the old `bison`.

FYI - an updated `bison` build got me going again on `Icarus`, `Verilator` and `Yosys` (the build that actually complained)

steveicarus, 2022-03-06:
A homebrew package already exists, and it is up to the package maintainer to handle that.

https://formulae.brew.sh/formula/icarus-verilog#default

closed 2022-03-06 by steveicarus