steveicarus/iverilog #552: Compatibility with other HDL tools (Quartus, ModelSim, and Vivado)

ranggihwang, 2021-09-30:
I found some code snippets that are fully compiled in Icarus Verilog, but not on other platforms.

On other platforms,
bit-wise NAND and NOR gate should be like ~(A&B) and ~(A|B)

but it Icarus verilog,
A~&B and A~|B work. but both of them are reduction operators (unary).
 
Aren't the problems should be edited in Icarus Verilog?

Moreover, it doesn't show any error when using =+ operator in Verilog, which is only supported in System Veilrog.

martinwhitaker, 2021-10-04:
The bitwise NAND, NOR, and XNOR operators are an Icarus-specific language extension. I thought this was documented, but can't find it said anywhere except in the test suite. As this is invalid syntax in regular Verilog, so can't give incorrect behaviour for valid Verilog code, there is currently no way of disabling this extension.

The compiler is fairly inconsistent about whether it accepts without warning, warns about, or errors out when encountering features that aren't available in the selected language generation.

caryr, 2021-10-05:
It should honor keywords based on the language flag, though even that is often not what you want since it would be better to report an appropriate error/warning message if a construct is not available in the specified generation after it has been parsed correctly.

steveicarus, 2022-03-06:
Maybe it would be good to disable this option with the -gno-icarus-misc flag. That flag is intended to disable Icarus Verilog extensions, and sure enough, this is an Icarus Verilog extension that slips through.

martinwhitaker, 2024-01-08:
Following Steve's comment, I'll label this as a bug so it has a chance of being worked on.

martinwhitaker, 2024-01-28:
Correcting myself - the binary XNOR operator is standard Verilog.

martinwhitaker, 2024-01-29:
Fixed in the master branch.

closed 2024-01-29 by martinwhitaker