steveicarus/iverilog #553: Internal assertion error when constructing multiple instances

mcandre, 2021-10-04:
Hi,

I am trying to implement an N-bit fulladder. However, when I tell iverilog to compile my design, then I get an internal assertion error.

Trace:

```console
$ iverilog -o bin/wholeass -g2012 test_wholeass.sv -Wall
ivl: elab_scope.cc:1450: void PGModule::elaborate_scope_mod_instances_(Design*, Module*, NetScope*) const: Assertion `(msb == 0) || (lsb != 0)' failed.
Aborted
```

Source:

https://github.com/mcandre/toys/tree/13d507a19dc3ffdd7c5a46b4b36dc4da4e6ff676/rtl/systemverilog/wholeass

Icarus Verilog version 12.0 (devel) (s20150603-1148-gef01dd1e), on Ubuntu 20.04 Focal Fossa.

Am I doing something wrong?

martinwhitaker, 2021-10-04:
You are using a SystemVerilog feature that isn't properly supported. Work around this by replacing the C style array declaration
````
    wholeass wa[N];
````
with the traditional Verilog style array declaration
````
    wholeass wa[N-1:0];
````
This is of course a compiler bug - if it doesn't support something it should output a proper error message to say so.

mcandre, 2021-10-05:
Agreed, my SystemVerilog syntax was bad.

I eventually found some working syntax, at least for Verilator:

https://github.com/mcandre/toys/blob/master/rtl/systemverilog/wholeass/test_wholeass.sv

A compiler parse error message would be most helpful, so that I will know better next time )

martinwhitaker, 2021-11-06:
Fixed in both the master and v11 branches.

closed 2021-11-06 by martinwhitaker