steveicarus/iverilog #554: Internal assertion error only when compiling with -g2012/-g2009/-g2005-sv

thebarnable, 2021-10-12:
I have a module that compiles without problems when using Verilog standards older than 2005, but runs into an internal assertion error when using newer versions.

Example code:
```
module TestModule #(parameter integer TEST_PARAM = 1) ();

genvar i;
generate
for (i=0; i<TEST_PARAM+1; i=i+1) begin : gen
    wire foo = 1;
    wire bar = (i==TEST_PARAM) ? 0 : gen[i+1].foo;  // this line induces the error
end
endgenerate

endmodule
```
Compiles without problems with: `iverilog -i TestModule.v` 

Errors when compiling with: `iverilog -g2012 -i TestModule.v`  (or -g2009/-g2005-sv)
```
ivl: symbol_search.cc:65: bool symbol_search(const LineInfo*, Design*, NetScope*, pform_name_t, symbol_search_results*, NetScope*): Assertion `scope' failed.
Aborted (core dumped)
```

My iverilog version is: Icarus Verilog version 11.0 (stable) (v11_0-82-g5ea6ee13)

SherilynSpika1988, 2021-11-03:
Same thing here

martinwhitaker, 2024-01-07:
This is fixed in the current stable release (v12), so closing.

closed 2024-01-07 by martinwhitaker