steveicarus/iverilog #556: Segmentation fault during OOB access to real array

TheZoq2, 2021-11-02:
Hi, thanks for this very useful software! I encountered a strange segmentation fault which I have narrowed down to this code:

```verilog
module  interp1d_tb ();
    reg clk;

    initial begin
        clk = 0;

        forever begin
            #1;
            clk = ~clk;
        end
    end

    real lut_values[1:0];
    assign lut_values[0] = 1;
    assign lut_values[1] = 2;

    wire[8-1:0] lut2_addr;
    reg[8-1:0] lut2_out;

    always @(posedge clk) begin
        lut2_out <= lut_values[lut2_addr];
    end

    // Check output
    always begin
        #1 // Must be at least 1

        $finish();
    end

    // Stage 3 (LUT)
    assign lut2_addr = 3; 
endmodule
```

when run using
```
iverilog -g2012 segfault_tb.sv -o build/buggy.vvp && ./build/buggy.vvp
```

This is about as minimal of an example as I can create. It looks like this is
caused by indexing out of bounds in the `lut_values` array which wasn't very
obvious in a bigger example.

The bug is not triggered if `lut2_addr` is `2` which is still OOB. In that case, I get
```
vvp: array.cc:730: double __vpiArray::get_word_r(unsigned int): Assertion `vsig' failed.
zsh: abort (core dumped)  ./build/buggy.vvp
```
Moving the 3 directly to the correctly gives an OOB warning and no segmentation fault

Changing the type of the `lut_values` array to `integer` gives no warnings, and
no assertion errors.

This is with icarus and vvp version
```
Icarus Verilog version 11.0 (stable) (v11_0)
```
from the arch repository.

I will see if I can reproduce this in the git version when I get more time this week

Github does not seem to allow uploads of core dumps. If that would be useful, let me know if I should upload it.

TheZoq2, 2021-11-03:
Update: I tried it on the git version and an still receiving a segfault, though this time it seems to be triggered on exit since it prints
```
>> iverilog -g2012 segfault_tb.sv -o build/buggy.vvp && ./build/buggy.vvp
segfault_tb.sv:29: $finish called at 1 (1s)
zsh: segmentation fault (core dumped)  ./build/buggy.vvp
```

Version string:
```
Icarus Verilog version 12.0 (devel) (s20150603-1149-ge3c22348)
```

martinwhitaker, 2021-11-03:
This bug only affects an array of real wires. If you replace the continuous assignments to the array elements with procedural assignments, there is no fault.

With a delay of 1 before the $finish, there is a race between that and the clock edge. The simulator completes the time step containing the call to $finish before exiting, hence the segmentation fault can occur after the finish message.

TheZoq2, 2021-11-03:
Thanks for  the reply, that does seem to have mitigated the issue

martinwhitaker, 2021-11-04:
Fixed in both the master and v11 branches.

closed 2021-11-04 by martinwhitaker

TheZoq2, 2021-11-05:
Thanks for the quick response and fix :heart:  Looks solved on my end as well